govorox/SSLClient #1: Release this library

jendakol, 2021-03-24:
Hi @govorox,
I've spent a lot of time searching for some working solution on connecting over SSL to web and MQTT with my ESP32 and SIM800L (LILIGO T-Call in particular). Your library is the only one that has actually worked for me! Would you be so kind to release it somehow so it's available e.g. at https://platformio.org/lib and everyone could find and use it easily?
I'm willing to help if you need, I just don't know how :smile:

govorox, 2021-03-25:
Hey! I've registered the library with PlatformIO. You can now find it by searching for SSLClient. It's at the bottom of the list at the moment 🙂

jendakol, 2021-03-25:
:heart:
I'll try to add some examples here once I have a time :smiley:

closed 2021-03-25 by jendakol