govorox/SSLClient #2: W5500 + ESP32 not working

fuxionlab, 2021-05-12:
This is a great library than other sslclient ive seen so far, but i couldn't make it work on a simple insecure mode MBEDTLS_SSL_VERIFY_NONE. Here is the sample code.
`SSLClient secure(&api_client);
void test_ssl(){
    if (!secure.connect("google.com", 443,NULL,NULL,NULL)) {
        Serial.println("connection failed");
    }  
}`
Its just a quicktest and im getting this error.
[E][ssl_client.cpp:35] _handle_error(): [start_ssl_client():280]: (-1) UNKNOWN ERROR CODE (0001)
[E][SSLClient2.cpp:110] connect(): start_ssl_client: -1
Im using W5500 ethernet on esp32 and as I understand on your ssl implementation its just a sslclient wrapper to any client instance.

Kromtar, 2021-05-29:
I don't know if it works for you but I used the [example ](https://github.com/govorox/SSLClient/blob/master/examples/Esp32/https_eth_LilyGo-T-ETH-POE/https_eth_LilyGo-T-ETH-POE.ino)(which uses ethernet) with the hostname "google.com" and I got the error 301. Then I modify the hostname to "www.google.com" and it works well.

govorox, 2021-06-11:
Thanks @Kromtar, the example above works well

closed 2021-06-11 by govorox

kapyaar, 2023-07-28:
@govorox @Kromtar Any chance you can include an example for Ethernet.h library? ie, using standard library (Ethernet.h)? 

I tried the way https://github.com/OPEnSLab-OSU/SSLClient example was shown, but they use Trust_anchors, and not certs.

Any change something along the lines of 

EthernetClient eth_transport_layer;
SSLClient client(&eth_transport_layer);

is available?

RobertByrnes, 2023-07-30:
Hi @kapyaar please can you open this as a new issue and label as a feature request. Or if you have already solved then please open a PR with the new example. Cheers.