govorox/SSLClient #6: Does this work with Secure MQTT?

gsvitak, 2021-05-30:
Hello

Thank you for the contributions!!! If this works, so exciting!

One question, the examples are for HTTP. Will this work with Secure MQTT (MQTTS) with PubSubClient?

Greg

Kromtar, 2021-05-30:
Hi

I tried it this morning...works fine :)
I created an example for you, look at [my frok](https://github.com/Kromtar/SSLClient/tree/master/examples/Esp32/mqtt_secure_gsm_SIM7000), i still haven't done the PR.

Test using: SIM7000G ([board](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G)), [this](https://github.com/knolleary/pubsubclient) MQTT Client and [this ](https://www.emqx.io/mqtt/public-mqtt5-broker)MQTT-Secure broker (with CA certificate)

reinaldolabreu, 2021-05-31:
Hi @Kromtar,  many thanks for your contribution. I used your code to do adapt a code for SIM800L with Secure MQTT x509 for Azure Iot Hub with telemetry and subscribe on Device Twin topic, it works! Follow the code.

[mqtt_gsm_SIM800L_Azure_x509_Device_Twin.zip](https://github.com/govorox/SSLClient/files/6571633/mqtt_gsm_SIM800L_Azure_x509_Device_Twin.zip)

Reinaldo Abreu

Kromtar, 2021-05-31:
@reinaldolabreu Thank you ! 
do you want me to add your example to this repository?

reinaldolabreu, 2021-06-03:
Please you can add.

> @reinaldolabreu Thank you !
> do you want me to add your example to this repository?

govorox, 2021-06-11:
Thank you @reinaldolabreu , I've added your code to examples

closed 2021-06-11 by govorox