govorox/SSLClient #8: Trying to connect to AWS

msentuerk, 2021-07-21:
Hello,

i am trying to connect to AWS. I am using the actual Azure example with some change.
I am also Using the Llillygo ESP32 Board  with the SIM800L.

`
#include <Arduino.h>
#include "utilities.h"
#include "ca_cert.h"
#include <Wire.h>
#include "SSLClient.h"

//MQTT Client lib: https://github.com/knolleary/pubsubclient
#include <PubSubClient.h>

// ESP32 LilyGo-T-Call-SIM800 SIM800L_IP5306_VERSION_20190610 (v1.3) pins definition
#define MODEM_RST 5
#define MODEM_PWRKEY 4
#define MODEM_POWER_ON 23
#define MODEM_TX 27
#define MODEM_RX 26
#define I2C_SDA 21
#define I2C_SCL 22
#define LED_GPIO 13
#define LED_PIN 13
#define LED_ON HIGH
#define LED_OFF LOW
#define IP5306_ADDR 0x75
#define IP5306_REG_SYS_CTL0 0x00

// Set serial for debug console (to the Serial Monitor)
#define SerialMon Serial
// Set serial for AT commands (to the SIM800 module)
#define SerialAT Serial1

#define TINY_GSM_DEBUG SerialMon
//#define DUMP_AT_COMMANDS          //Uncomment to see AT commands on Serial

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800   // Modem is SIM800
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb

// Include after TinyGSM definitions
#include <TinyGsmClient.h>

// Your GPRS credentials (leave empty, if missing)
const char apn[] = "internet.t-mobile";       // Your APN
const char gprs_user[] = "t-mobile"; // User
const char gprs_pass[] = "tm"; // Password
const char simPIN[] = "";    // SIM card PIN code, if any

// MQTT Config

const char mqtt_broker[] = "a3370efcvjelqg-ats.iot.us-west-2.amazonaws.com";
#define MQTT_DEVICE     "ESP2"

int secure_port = 8883; // TCP-TLS Port

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm sim_modem(debugger);
#else
TinyGsm sim_modem(SerialAT);
#endif

//Layers stack
//TinyGsm sim_modem(SerialAT);
TinyGsmClient gsm_transpor_layer(sim_modem);
SSLClient secure_presentation_layer(&gsm_transpor_layer);
PubSubClient client(secure_presentation_layer);

// Power configuration for SIM800L_IP5306_VERSION_20190610 (v1.3) board
bool setupPMU()
{
  bool en = true;
  Wire.begin(I2C_SDA, I2C_SCL);
  Wire.beginTransmission(IP5306_ADDR);
  Wire.write(IP5306_REG_SYS_CTL0);
  if (en)
  {
    Wire.write(0x37);
  }
  else
  {
    Wire.write(0x35);
  }
  return Wire.endTransmission() == 0;
}

// For read the MQTT events
void callback(char *topic, byte *payload, unsigned int length)
{
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++)
  {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

// To connect to the broker
void reconnect()
{
  // Loop until we're reconnected
  while (!client.connected())
  {
    Serial.println("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect(MQTT_DEVICE))
    //if (client.connect(client_name))
    {
      Serial.println("-----------------------------------connected-----------------------");
      // Once connected, publish an announcement...
     // client.publish("outTopic", "hello world");
      client.publish("esp32/sub","Test");     // Topic to publish telemetry

      // ... and resubscribe
      client.subscribe("esp32/sub");                                //Topic to subscribe Device Twin
    }
    else
    {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println("...try again in 5 seconds");
      delay(5000);
    }
  }
}

// Modem initial setup (cold start)
void setupModem()
{
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_PWRKEY, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  pinMode(LED_PIN, OUTPUT);
  
  // Reset pin high
  digitalWrite(MODEM_RST, HIGH);

  // Turn on the Modem power first
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Pull down PWRKEY for more than 1 second according to manual requirements
  digitalWrite(MODEM_PWRKEY, HIGH);
  delay(200);
  digitalWrite(MODEM_PWRKEY, LOW);
  delay(1200);
  digitalWrite(MODEM_PWRKEY, HIGH);

  // Initialize the indicator as an output
  digitalWrite(LED_PIN, LOW);
}

void setup()
{
  SerialMon.begin(115200);
  delay(10);

  // Start power management
  if (!setupPMU())
  {
    Serial.println("Setting power error");
  }

  // Set SIM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);

  
  //Add CA Certificate
  secure_presentation_layer.setCACert(AWS_CERT_CA);
  secure_presentation_layer.setCertificate(AWS_CERT_CRT);    //x509 client Certificate
  secure_presentation_layer.setPrivateKey(AWS_CERT_PRIVATE);      //x509 client key

  // Modem initial setup
  setupModem();

  // MQTT init
  client.setServer(mqtt_broker, secure_port);
  client.setCallback(callback);
}

void loop()
{
  SerialMon.print("Initializing modem...");
  if (!sim_modem.init())
  {
    SerialMon.print(" fail... restarting modem...");
    setupModem();
    // Restart takes quite some time
    // Use modem.init() if you don't need the complete restart
    if (!sim_modem.restart())
    {
      SerialMon.println(" fail... even after restart");
      return;
    }
  }
  SerialMon.println(" OK");
  
  // General information
  String name = sim_modem.getModemName();
  Serial.println("Modem Name: " + name);
  String modem_info = sim_modem.getModemInfo();
  Serial.println("Modem Info: " + modem_info);

  // Unlock your SIM card with a PIN if needed
  if (strlen(simPIN) && sim_modem.getSimStatus() != 3)
  {
    sim_modem.simUnlock(simPIN);
  }

  // Wait for network availability
  SerialMon.print("Waiting for network...");
  if (!sim_modem.waitForNetwork(240000L))
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");
  
  // Connect to the GPRS network
  SerialMon.print("Connecting to network...");
  if (!sim_modem.isNetworkConnected())
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  // Connect to APN
  SerialMon.print(F("Connecting to APN: "));
  SerialMon.print(apn);
  if (!sim_modem.gprsConnect(apn, gprs_user, gprs_pass))
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  digitalWrite(LED_PIN, HIGH);
  SerialMon.println(" OK");

  // More info..
  Serial.println("");
  String ccid = sim_modem.getSimCCID();
  Serial.println("CCID: " + ccid);
  String imei = sim_modem.getIMEI();
  Serial.println("IMEI: " + imei);
  String cop = sim_modem.getOperator();
  Serial.println("Operator: " + cop);
  IPAddress local = sim_modem.localIP();
  Serial.println("Local IP: " + String(local));
  int csq = sim_modem.getSignalQuality();
  Serial.println("Signal quality: " + String(csq));

  // MQTT Test loop
  // As long as we have connectivity
  while (sim_modem.isGprsConnected())
  {
    // We maintain connectivity with the broker
    if (!client.connected())
    {
      reconnect();
    }
    // We are listening to the events
    client.loop();
  }
  
  // Disconnect GPRS and PowerOff
  // Apparently the "gprsDisconnect()" method (TinyGSM) are not working well with the SIM7000...
  // ...you have to use additionally "poweroff()".
  // With that, the modem can be connected again in the next cycle of the loop.
  //sim_modem.gprsDisconnect();
  //sim_modem.poweroff();

  delay(15000);

}
`

The error I am getting ist this:

> [E][SSLClient.cpp:110] connect(): start_ssl_client: -1
failed, rc=-2...try again in 5 seconds

My full log logs like this:
`
Initializing modem...[1440] ### TinyGSM Version: 0.11.4
[1440] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
[3553] ### Modem: SIMCOM SIM800L
[3553] ### Modem: SIMCOM SIM800L
 OK
[6585] ### Modem: SIMCOM SIM800L
Modem Name: SIMCOM SIM800L
Modem Info: SIM800 R14.18
Waiting for network... OK
Connecting to network... OK
Connecting to APN: internet.t-mobile OK

CCID: YYYXXXXX
IMEI: XXXXXAAAA
Operator: D1
Local IP: 11111111
Signal quality: 27
Attempting MQTT connection...
[E][SSLClient.cpp:110] connect(): start_ssl_client: -1
failed, rc=-2...try again in 5 seconds
Attempting MQTT connection...
`

msentuerk, 2021-07-25:
Problem solved! The sim card was the problem. Switched to another provider.

closed 2021-07-25 by msentuerk

ashish-iottive, 2021-08-13:
Hey @msentuerk  could you provide Full Sketch including `ca_cert.h` File and `utilities.h` content So that I can find my mistake, or you can provide Full Code in zip file with removing your  AWS Endpoint, ROOTCA Private Key and Device Certificate content.
Thanks and Regards