govorox/SSLClient #13: Keep getting start_ssl_client: -1

Lavoisier91, 2021-11-18:
```
16:18:53.433 -> Connecting to xxx.com[259775][D][SSLClient.cpp:103] connect(): Connecting to xxx.com:443
16:18:53.433 -> [259775][V][ssl_client.cpp:156] start_ssl_client(): Free internal heap before TLS 315972
16:18:53.433 -> [259782][D][ssl_client.cpp:158] start_ssl_client(): Connecting to xxx.com:443
16:18:53.901 -> [260262][V][ssl_client.cpp:172] start_ssl_client(): Seeding the random number generator
16:18:53.901 -> [260263][V][ssl_client.cpp:181] start_ssl_client(): Setting up the SSL/TLS structure...
16:18:53.901 -> [260266][V][ssl_client.cpp:194] start_ssl_client(): Loading CA cert
16:18:53.995 -> [260336][V][ssl_client.cpp:259] start_ssl_client(): Setting hostname for TLS session...
16:18:53.995 -> [260337][V][ssl_client.cpp:272] start_ssl_client(): Setting up IO callbacks...
16:18:53.995 -> [260340][V][ssl_client.cpp:276] start_ssl_client(): Performing the SSL/TLS handshake...
16:18:54.370 -> [260742][D][ssl_client.cpp:136] client_net_send(): SSL client TX res=303 len=303
16:18:59.098 -> [265461][E][SSLClient.cpp:110] connect(): start_ssl_client: -1
16:18:59.098 -> [265461][V][ssl_client.cpp:330] stop_ssl_socket(): Cleaning SSL connection.
16:18:59.144 ->  fail
```

What I can not find a solution to is this:
`[ 43795][E][SSLClient.cpp:110] connect(): start_ssl_client: -1`

Which I think is coming from ssl_client.cpp @151
`log_e("Client provider not initialised");`

So, what am I missing here?

andersruneson, 2023-06-11:
Could be the timeout that is default set to 0.
I had an similar issue and should be solved by setting the timeout:
`sslClient.setHandshakeTimeout(3);`

govorox, 2023-06-11:
are you able to fix and create a PR? thanks

andersruneson, 2023-06-11:
I can try. But it looks like the timeout is initialized to 120 s.
`sslclient->handshake_timeout = 120000;`

I just saw the issue myself and will test if setting the timeout helps but I saw this in the log:
`[ssl_client.cpp:98] client_net_recv_timeout(): SSL client RX (received=-1 expected=5 in 0ms)`
And 0ms indicates an extremely short timeout.

PR #41 (2023-07-24): Fix stop() function calls to stop_ssl_socket()
UPDATE: @govorox Your work creating and publishing this library is obviously appreciated by the community. If mainting this is no longer an option for you consider adding another maintainer  or more. As I use this library as a dependency I would be happy to maintain it and address Issues and PR either solo or as one of two or more maintainers.

PULL REQUEST:
When running on an ESP32 I noticed that I was getting `abort()` called due to an exception starting in a connect function. The throw was a null pointer exception when the client had already te [... cut]