govorox/SSLClient #17: TLS 1.3 support

chupachupbum, 2022-06-24:
It seems that SSLClient doesn't work with sites have TLS 1.3. I tried ngrok and drive.google, both use TLS 1.3 and it doesn't work; for sites are still using TLS 1.2 like httpbin.org, things go fine. MbedTLS already has new version which support TLS 1.3, so I wonder if you can update your module with the new Mbed-TLS version?

genotix, 2022-06-24:
Hi, I’m not actively maintaining this project and shared it so people could benefit from it.

Feel free to clone it and modify anything.

chupachupbum, 2022-08-04:
I figured out that it only doesn't work for some websites, ironically 2 of them are those I tested, ngrok and google drive. For other hosts using TLS 1.3 it's still working fine. Yet I haven't found what cause the problem or the different between the working and not working ones. 
Will close this for now.

closed 2022-08-04 by chupachupbum