govorox/SSLClient #22: fatal error: mbedtls/net.h: No such file or directory

inuitviking, 2022-08-18:
Hi

I am trying to connect an Arduino MKR WiFi 1010 to a secure MQTT broker (Mosquitto).
I found this library in the search for secure connection with CA, client certificate, and client key.

When I try to use it and compile it, I get the following error:
```
In file included from .pio/libdeps/mkrwifi1010/SSLClient/src/SSLClient.h:23:0,
                 from src/main.cpp:5:
.pio/libdeps/mkrwifi1010/SSLClient/src/ssl_client.h:9:10: fatal error: mbedtls/net.h: No such file or directory
 #include "mbedtls/net.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
```

I have also tried including the mbedtls library by putting "https://github.com/Mbed-TLS/mbedtls.git" in the platformio.ini file.
When I search for the "net.h" file in mbedtls, I can't find it, but I do find a sockets.h. 

I see this is an active project, how have you guys fixed this issue?

TheRedSpy15, 2022-10-13:
Also having similar issue with mbedtls/platform.h

TheRedSpy15, 2022-10-17:
@kiddtang I'm actually following your tutorial here: https://dogcomp.medium.com/send-https-request-from-sim-7600x-lte-module-4f76be19e900

And ended up with this issue

govorox, 2022-10-18:
The module uses MBEDTLS library. You need to make sure the it is included in the build.
Some platform may not have ported it yet, or do not include it by default.

inuitviking, 2022-10-18:
Hi, I completely forgot about this issue.

I ended up using [this](https://github.com/OPEnSLab-OSU/SSLClient) other library instead.
As this was for a school project, and borrowed the MKR WiFi 1010, I unfortunately do not have a capable board to debug this on.

Should I close the issue or let it be open for others?

RobertByrnes, 2023-07-25:
Hi @inuitviking I am closing this issue as I believe this is now fixed in a subsequent PR. Please open a new issue if this is still of concern. Thank you.

closed 2023-07-25 by RobertByrnes

suntorytimed, 2023-08-10:
Just for your information, the issue is that mbedtls removed those deprecated headers in major version 3 of the library. mbedtls 2.28 is still supported until end to 2024.

woodlist, 2024-04-24:
> The module uses MBEDTLS library. You need to make sure that it is included in the build. Some platforms may not have ported it yet, or do not include it by default.

Please, explain how to install the MBEDTLS.

woodlist, 2024-04-24:
> Hi
>
> I am trying to connect an Arduino MKR WiFi 1010 to a secure MQTT broker (Mosquitto). I found this library in the search for secure connection with CA, client certificate, and client key.
>
> When I [...]

[https://github.com/search?q=repo%3AMbed-TLS%2Fmbedtls%20net.c&type=code](url)
So do just rename the `#include "mbedtls/net.h"` to `#include "mbedtls/net_sockets.h`

RobertByrnes, 2024-04-24:
Hi @suntorytimed you might try something like this first:

`lib_deps =
    https://github.com/Mbed-TLS/mbedtls.git@2.28`
 
 This is based on the comment above and I haven't tested that as yet but:
 
 "Just for your information, the issue is that mbedtls removed those deprecated headers in major version 3 of the library. mbedtls 2.28 is still supported until end to 2024."
 
 This library needs updating to use the latest version of mbedtls.....