govorox/SSLClient #24: Designed and tested on ESP32, but should be portable to other platforms.

bvernham, 2022-09-02:
How is this portable to other platforms as this is dependent on mbedtls?

Thanks

Bruce

govorox, 2022-09-02:
This was meant to fill a gap for one of my ESP32 based projects.
Never claimed it was fully portable, hence 'should be' in the statement.
Naturally, it works on ARM as well as ESP32, although I haven't tested it on the newer RISCV variants
Here is some info about **mbedtls** portability: https://mbed-tls.readthedocs.io/en/latest/kb/how-to/how-do-i-port-mbed-tls-to-a-new-environment-OS/

RobertByrnes, 2023-07-25:
Hi @bvernham Will now close this issue as is inactive - have marked as documentation purely due to the link above to mbed tls docs

closed 2023-07-25 by RobertByrnes