govorox/SSLClient #39: I am getting heavy RAM consumption while using SSLClient

Thermelgy-Repo, 2023-06-23:
Dear @govorox, @moo-im-a-cow , @Kromtar, @ittacco, @kiddtang,

Thank you so much for the efforts for this great library.

I am getting heavy RAM consumption while using our SSLClient, consuming Almost 40KB of precious SRAM. Any observation?. Any tips to decrease the RAM consumption

ifelse-code, 2023-09-14:
I am also with this "problem".

RobertByrnes, 2023-09-17:
Hey @Thermelgy-Repo @ifelse-code - do either of you have any profile on this: presumably the ram your seeing consumed is DRAM rather than PSRAM. If it is then :

- esp_get_free_heap_size()
- esp_get_minimum_free_heap_size()
- heap_caps_get_free_size()
- heap_caps_get_largest_free_block()

Throwing some debug lines in with these funcs. might be helpful?

What size cert are you using?

Try adjusting buffer size? SSL_CLIENT_SEND_BUFFER_SIZE ssl_client.h:23

RobertByrnes, 2023-11-20:
I am closing this issue due to no recent activity. Please open a new issue if the problem persists.

closed 2023-11-20 by RobertByrnes