govorox/SSLClient #42: ESP32 resets when the CA certificate is invalid for MQTTS connection using PubSubClient !!!

winnergeorge, 2023-07-23:
This is a great library for adding SSL functionality to WiFi, Ethernet (W5500 based) or Simcom 4G module (76XX based). I have tested Secure MQTT on WiFi, Ethernet and 4G LTE and it works. However if the CA certificate is invalid, ESP32 resets when mqttClient.connect("ClientID");  (this function is from PubSubClient library) line is executed. 
ESP resets and
"core 1 panic'ed (loadprohibited). exception was unhandled" message is thrown. Also EXCVADDR: 0x00000000.

Any suggestions as in how to handle this exception? The worst scenario in this case should be that the Server refuses connection, not that ESP resets!

RobertByrnes, 2023-07-23:
Hi, I think you'll find pull request #41 amd also #40 both address this issue. Sadly, I think the maintainer doesn't have time for this repo any longer.  I have suggested allowing othwr maintainers, perhaps.you could do the same. Mybe try the fix nd see if works for you then comment on the PR that is closes your issue if it does :)

winnergeorge, 2023-07-24:
Hi Robert, your PR fixes this issue perfectly. No more ESP32 crash :)

closed 2023-07-24 by winnergeorge