govorox/SSLClient #43: New version published to PlatformIO

RobertByrnes, 2023-07-25:
Hi @govorox I have published a new version to PlatformIO as digitaldragon/SSLCLient. I tried to use the author as govorox/SSLCLient but was unable to publish to an account I do not own.

I do not currently know of a work around for this but will leave this note here. If anyone does please comment...

govorox, 2023-07-25:
Thanks @RobertByrnes, I don't mind how it is being published. Is there anything I need to do on my part?

RobertByrnes, 2023-07-25:
@govorox as long as you don't mind the account name being different there is nothing you need to do - it will still all point back to this repository. Cheers...

govorox, 2023-07-25:
I don't mind at all, and thank you for putting all the work in, and going through the outstanding issues!

closed 2023-08-01 by RobertByrnes