govorox/SSLClient #45: client_net_send - Buffer issue when writing data larger than the receiving buffer

deeja, 2023-08-14:
https://github.com/govorox/SSLClient/blob/d7cd4afeb10b0ce1c7a8c6c440d9a2cfaf473310/src/ssl_client.cpp#L134

If the buffer is smaller than the data being written, then the call fails.

My fix (although ugly) is this 
```cpp
 // Sending data in chunks using a small buffer
    const int bufferSize = 1024; // Size of the buffer I'm writing to
    int result;
    for (int i = 0; i < len; i += bufferSize)
    {
        int bytesToWrite;

        if (bufferSize > len - i)
        {
            bytesToWrite = len - i;
        }
        else
        {
            bytesToWrite = bufferSize;
        }

        // Create a new buffer for each chunk
        unsigned char buffer[bytesToWrite];
        memcpy(buffer, &buf[i], bytesToWrite);

        // Send the buffer to the client
        result = client->write(buffer, bytesToWrite);
        if (result == 0)
        {
            log_e("write failed");
            result = MBEDTLS_ERR_NET_SEND_FAILED;
            break;
        }
    }
```

RobertByrnes, 2023-08-14:
Hi @deeja Thanks for submitting this. It would be great to see this turn into a pull request to have it added to the repo. But first, would you please share a bit more around the issue. What is the Hardware, the platform (e.g. Arduino / PlatformIO / ESP_IDF that you used. Was this the only code change you implemented in making a fix? What was the exact error that came up? It would helpful if you could share that.

deeja, 2023-08-15:
Hey @RobertByrnes  
Arduino on PlatformIO 
Hardware: Onboard Lilygo A7670 / SIM7000 (WROVER ESP32)
The TinyGsm RX Buffer is 1024 bytes so I'm guessing the TX is the same? https://github.com/Xinyuan-LilyGO/T-A7670X/blob/69c6275bfc2d6c1e233e5c1c7cfea5dca57e29f4/examples/ATdebug/ATdebug.ino#L8
Would create a PR, but honestly my C++ isn't up to scratch.

RobertByrnes, 2023-08-22:
Hi @deeja try this implementation your fix please and let me know how you get on - [fix](https://github.com/govorox/SSLClient/tree/fix-buffer-issue-when-writing-data-larger-than-the-receiving-buffer)

RobertByrnes, 2023-08-22:
[Pull Request #46](https://github.com/govorox/SSLClient/pull/46)

deeja, 2023-08-26:
Hey I'll check it out when the new devices arrive in a couple of weeks

RobertByrnes, 2023-08-28:
HI @deeja try pulling the branch again - have just updated. I get an SSL connection OK with this code now...

deeja, 2023-08-29:
Yeah, that looks to be working now. No issues with the connection. Thanks!

PR #46 (2023-09-17): Fix buffer issue when writing data larger than the receiving buffer
Closes #45
closed 2023-09-17 by RobertByrnes