govorox/SSLClient #53: What root certificate to load for a generic use of SSLClient

Bascy, 2023-09-20:
In our ESP32 framework we have an function that is used for sending generic HTTP and HTTPS requests, we have similar implementations for Ethernet connections and for connections via a Sim7600  gsm module.

To actually use secure communication in case of an https request, I should set a CAcertificate .. but what certificate should I set if the server in the request can vary?

I'm not very  skilled in secure communications, so any suggestions are welcome

```
typedef std::function <int(HttpClient&)> HttpClientFunction;

int EthernetNetworkProvider::runWithHttpClient(const URL& url, HttpClientFunction func) {
  EthernetClient ethernetHttpClient;

  int result = 0;

  if (url.getProtocol() == "https://") {
    SSLClient client(&ethernetHttpClient);
    client.setTimeout(10);
    client.setCACert(letsencrypt_root_ca);  // Which certificate to load here?
    HttpClient http{client, url.getServer().c_str(), url.getPort()};
    http.connectionKeepAlive();
    result = func(http);
    http.stop();
  } else {
    HttpClient http{ethernetHttpClient, url.getServer().c_str(), url.getPort()};
    http.connectionKeepAlive();
    result = func(http);
    http.stop();
  }

  ethernetHttpClient.stop();
  return result;
}
```

RobertByrnes, 2023-09-20:
I would look for a common root certificate and use that or if there is no single common root then create a structure which contains multiple certs and can try different ones until finding the correct one. E.g lets encrypt free certs have a common root cert. Beware of memory consumption...

Bascy, 2023-09-20:
Ok, so essentially trial and error ;)
Given a certificate and a url, what is the fastest way to check if they match?

RobertByrnes, 2023-09-23:
Click on the padlock in browser address bar for website and have a look at the cert chain - follow it back a step or two to the root cert

closed 2023-10-16 by Bascy