govorox/SSLClient #55: SSL - No RNG was provided to the SSL module

der-elektrotechniker, 2023-09-24:
Hi!

Many thanks for this library.

I want to connect to a MQTT-Server wirht authentification and client-certificate.

I'm trying to use your example code for this purpose. The certificates are all copied into the certs.h properly. The code compiles correctly. After flashing it to the Lilygo-T-SIM7000G with ESP32 on it, it connects to GSM network correctly, but there is a problem with the SSL-connection to the MQTT Server. The serial monitor shows the following error message:

```
Attempting MQTT connection...[ 18962][E][ssl_client.cpp:41] _handle_error(): [start_ssl_client():347]: (-29696) SSL - No RNG was provided to the SSL module
[ 18963][E][SSLClient.cpp:133] connect(): start_ssl_client: -29696

assert failed: heap_caps_free heap_caps.c:360 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: <anonymized>

ELF file SHA256: <anonymized>

Rebooting...
```

How can I add RNG to the SSL module?

Thank you very much in advance!

RobertByrnes, 2023-09-24:
Hi @der-elektrotechniker,

MBEDTLS_ERR_SSL_NO_RNG = -29696! This comes from the underlying mbedtls library. First suggestion is to try pulling the latest version of this lib 1.1.5 and running with log_v enabled. You will have a LOT of logging there but I expect it will help yo uunderstand where things are going wrong. It would be useful for you to share your code here, it would make it possible for others to stop errors if they exist and suggest fixes.

der-elektrotechniker, 2023-09-25:
Hi @RobertByrnes ,

thank you very much for your fast response!

I realized, that I was using version 1.1.4 of your library and that wasn't the latest. I don't know how that happened. I used PlatformIO's library manager to download it, and that was the latest version...

Now that I changed the version number to the latest in platformio.ini the error changed. It's not the RNG error anymore.

I put "build_flags = -DCORE_DEBUG_LEVEL=5" in my platformio.ini so that I have log_v enabled.

The output is now the following:

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[     2][V][ssl_client.cpp:202] ssl_init(): Init SSL
[    20][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   143][D][SSLClient.cpp:481] setCACert(): Set root CA
[   143][D][SSLClient.cpp:495] setCertificate(): Set client CA
[   144][D][SSLClient.cpp:510] setPrivateKey(): Set client PK
Initializing modem... OK
Modem Name: SIMCOM SIM7000G
Modem Info: SIM7000G R1529
Waiting for network... OK
Connecting to network... OK
Connecting to APN: <anonymized> OK

CCID: <anonymized>
IMEI: <anonymized>
Operator: <anonymized>
Local IP: <anonymized>
Signal quality: 26
[ 17092][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 17093][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
Attempting MQTT connection...[ 17106][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 17106][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 17113][V][SSLClient.cpp:161] connect(): connect with CA
[ 17118][V][SSLClient.cpp:219] connect(): Connecting to <anonymized>:8883
[ 17125][V][SSLClient.cpp:220] connect(): Timeout value: 0
[ 17131][V][SSLClient.cpp:221] connect(): CA Certificate: Provided
[ 17136][V][SSLClient.cpp:222] connect(): Client Certificate: Provided
[ 17143][V][SSLClient.cpp:223] connect(): Private Key: Provided
[ 17148][V][ssl_client.cpp:240] start_ssl_client(): Free internal heap before TLS 342060
[ 17156][V][ssl_client.cpp:241] start_ssl_client(): Connecting to <anonymized>:8883
[ 17165][V][ssl_client.cpp:260] start_ssl_client(): Client pointer: 0x3ffc2520
[ 17370][V][ssl_client.cpp:269] start_ssl_client(): Seeding the random number generator
[ 17370][V][ssl_client.cpp:271] start_ssl_client(): Entropy context initialized
[ 17374][V][ssl_client.cpp:280] start_ssl_client(): Random number generator seeded, ret: 0
[ 17381][V][ssl_client.cpp:283] start_ssl_client(): Setting up the SSL/TLS defaults...
[ 17390][V][ssl_client.cpp:293] start_ssl_client(): SSL config defaults set, ret: 0
[ 17396][V][ssl_client.cpp:297] start_ssl_client(): Loading CA cert
[ 17408][V][ssl_client.cpp:356] start_ssl_client(): Loading CRT cert
[ 17414][V][ssl_client.cpp:364] start_ssl_client(): Loading private key
[ 17424][V][ssl_client.cpp:380] start_ssl_client(): Setting hostname for TLS session...
[ 17424][V][ssl_client.cpp:398] start_ssl_client(): Setting up IO callbacks...
[ 17429][V][ssl_client.cpp:402] start_ssl_client(): Setting timeout to 0
[ 17436][V][ssl_client.cpp:406] start_ssl_client(): Performing the SSL/TLS handshake...
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400864e1  PS      : 0x00060230  A0      : <anonymized>  A1      : <anonymized>  
[... 307 lines cut ...]
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	knolleary/PubSubClient@^2.8
	vshymanskyy/TinyGSM@^0.11.7
	digitaldragon/SSLClient@^1.1.5
	armmbed/mbedtls@^2.23.0
monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
```

Many thanks for your help in advance!

RobertByrnes, 2023-09-25:
@der-elektrotechniker OK good one on detail. What add to this is if you put this line in your env in platform.ini: `monitor_filters = esp32_exception_decoder, log2file, colorize, debug, printable` 

It will save the serial output to a file and when it throws the unhandled exception on making the handshake you should get a stack trace back out of it - could show a little bit more...

der-elektrotechniker, 2023-09-25:
Wow, that was really fast! Thank you for your response, @RobertByrnes !

Here is that, what is written in the file:

```
Initializing modem... OK
Modem Name: SIMCOM SIM7000G
Modem Info: SIM7000G R1529
Waiting for network... OK
Connecting to network... OK
Connecting to APN: <anonymized> OK

CCID: <anonymized>
IMEI: <anonymized>
Operator: <anonymized>
Local IP: <anonymized>
Signal quality: 28
[ 17180][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 17181][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
Attempting MQTT connection...[ 17194][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 17194][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 17201][V][SSLClient.cpp:161] connect(): connect with CA
[ 17206][V][SSLClient.cpp:219] connect(): Connecting to <anonymized>:8883
[ 17213][V][SSLClient.cpp:220] connect(): Timeout value: 0
[ 17219][V][SSLClient.cpp:221] connect(): CA Certificate: Provided
[ 17224][V][SSLClient.cpp:222] connect(): Client Certificate: Provided
[ 17231][V][SSLClient.cpp:223] connect(): Private Key: Provided
[ 17236][V][ssl_client.cpp:238] start_ssl_client(): Free internal heap before TLS 342060
[ 17244][V][ssl_client.cpp:239] start_ssl_client(): Connecting to <anonymized>:8883
[ 17252][V][ssl_client.cpp:258] start_ssl_client(): Client pointer: 0x3ffc2520
[ 17451][V][ssl_client.cpp:267] start_ssl_client(): Seeding the random number generator
[ 17451][V][ssl_client.cpp:269] start_ssl_client(): Entropy context initialized
[ 17456][V][ssl_client.cpp:278] start_ssl_client(): Random number generator seeded, ret: 0
[ 17463][V][ssl_client.cpp:281] start_ssl_client(): Setting up the SSL/TLS defaults...
[ 17472][V][ssl_client.cpp:291] start_ssl_client(): SSL config defaults set, ret: 0
[ 17478][V][ssl_client.cpp:295] start_ssl_client(): Loading CA cert
[ 17490][V][ssl_client.cpp:354] start_ssl_client(): Loading CRT cert
[ 17495][V][ssl_client.cpp:362] start_ssl_client(): Loading private key
[ 17505][V][ssl_client.cpp:378] start_ssl_client(): Setting hostname for TLS session...
[ 17506][V][ssl_client.cpp:396] start_ssl_client(): Setting up IO callbacks...
[ 17511][V][ssl_client.cpp:400] start_ssl_client(): Setting timeout to 0
[ 17517][V][ssl_client.cpp:404] start_ssl_client(): Performing the SSL/TLS handshake...
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400864e1  PS      : 0x00060430  A0      : 0x80105f2c  A1      : 0x3ffb19c0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3f40c1c6  A12     : 0x00000000  A13     : 0x3ffb18b0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400864e1  LEND    : 0x400864f1  LCOUNT  : 0xffffffff  

Backtrace: 0x400864de:0x3ffb19c0 0x40105f29:0x3ffb19d0 0x4010aa96:0x3ffb1ce0 0x4010aad2:0x3ffb1d70 0x400d80d0:0x3ffb1db0 0x400e3864:0x3ffb1df0 0x400d5faf:0x3ffb1e70 0x400d4da0:0x3ffb20e0 0x400d5039:0x3ffb2120 0x400d44b1:0x3ffb2150 0x400d4606:0x3ffb2190 0x400d2582:0x3ffb21c0 0x400d3f3f:0x3ffb21e0 0x400d82bd:0x3ffb2290

  #0  0x400864de:0x3ffb19c0 in strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:82
  #1  0x40105f29:0x3ffb19d0 in _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528
  #2  0x4010aa96:0x3ffb1ce0 in _vsnprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:70 (discriminator 4)
  #3  0x4010aad2:0x3ffb1d70 in vsnprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
  #4  0x400d80d0:0x3ffb1db0 in log_printfv at /home/patrick/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:563
  #5  0x400e3864:0x3ffb1df0 in __wrap_log_printf at /Users/ficeto/Desktop/ESP32/ESP32S2/esp32-arduino-lib-builder/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:426
  #6  0x400d5faf:0x3ffb1e70 in start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*) at .pio/libdeps/esp32dev/SSLClient/src/ssl_client.cpp:425
  #7  0x400d4da0:0x3ffb20e0 in SSLClient::connect(char const*, unsigned short, char const*, char const*, char const*) at .pio/libdeps/esp32dev/SSLClient/src/SSLClient.cpp:230
  #8  0x400d5039:0x3ffb2120 in SSLClient::connect(char const*, unsigned short) at .pio/libdeps/esp32dev/SSLClient/src/SSLClient.cpp:162
  #9  0x400d44b1:0x3ffb2150 in PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*, bool) at .pio/libdeps/esp32dev/PubSubClient/src/PubSubClient.cpp:192
      (inlined by) PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*, bool) at .pio/libdeps/esp32dev/PubSubClient/src/PubSubClient.cpp:181
  #10 0x400d4606:0x3ffb2190 in PubSubClient::connect(char const*) at .pio/libdeps/esp32dev/PubSubClient/src/PubSubClient.cpp:166
  #11 0x400d2582:0x3ffb21c0 in reconnect() at src/main.cpp:84
  #12 0x400d3f3f:0x3ffb21e0 in loop() at src/main.cpp:251
[... cut ...]
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	knolleary/PubSubClient@^2.8
	vshymanskyy/TinyGSM@^0.11.7
	digitaldragon/SSLClient@^1.1.6
	armmbed/mbedtls@^2.23.0
monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
monitor_filters = esp32_exception_decoder, log2file, colorize, debug, printable
```

Thanks in advance!

der-elektrotechniker, 2023-09-25:
PS: Let me ask a question: Why isn't there any possibility to define an analog pin for the random generator? Isn't that necessary for really good random numbers? Or is that done automatically by some function in your library?

RobertByrnes, 2023-09-26:
- LoadProhibited generally means incorrect memory referenced.
- line 425 is the culprit. It tries to log `log_v("Protocol is %s Ciphersuite is %s", mbedtls_ssl_get_version(&ssl_client->ssl_ctx), mbedtls_ssl_get_ciphersuite(&ssl_client->ssl_ctx));`
- line 424 checks the certs for NULL so they are not NULL
- it is possible one of the 2 mbedtls func calls in the log statement is returning NULL or and broken pointer.

```
const char* ssl_version = mbedtls_ssl_get_version(&ssl_client->ssl_ctx);
const char* ssl_ciphersuite = mbedtls_ssl_get_ciphersuite(&ssl_client->ssl_ctx);

if (ssl_version) {
    log_v("Protocol is %s", ssl_version);
} else {
    log_e("Failed to get Protocol");
}
if (ssl_ciphersuite) {
    log_v("Ciphersuite is %s", ssl_ciphersuite);
} else {
    log_e("Failed to get Ciphersuite");
}
```

trying editing line 425 tp the following and see...

der-elektrotechniker, 2023-09-26:
Hi @RobertByrnes ,

thanks for your tips!

I replaced the line `log_v("Protocol is %s Ciphersuite is %s", mbedtls_ssl_get_version(&ssl_client->ssl_ctx), mbedtls_ssl_get_ciphersuite(&ssl_client->ssl_ctx));
` with the code block from your last answer.

Now the output is the following:

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[     2][V][ssl_client.cpp:200] ssl_init(): Init SSL
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   141][D][SSLClient.cpp:481] setCACert(): Set root CA
[   142][D][SSLClient.cpp:495] setCertificate(): Set client CA
[   142][D][SSLClient.cpp:510] setPrivateKey(): Set client PK
Initializing modem... OK
Modem Name: SIMCOM SIM7000G
Modem Info: SIM7000G R1529
Waiting for network... OK
Connecting to network... OK
Connecting to APN: <anonymized> OK

CCID: <anonymized>
IMEI: <anonymized>
Operator: <anonymized>
Local IP: <anonymized>
Signal quality: 26
[ 16949][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 16950][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
Attempting MQTT connection...[ 16963][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 16963][V][SSLClient.cpp:394] read(): This is the iClient->read() implementation
[ 16970][V][SSLClient.cpp:161] connect(): connect with CA
[ 16975][V][SSLClient.cpp:219] connect(): Connecting to <anonymized>:8883
[ 16982][V][SSLClient.cpp:220] connect(): Timeout value: 0
[ 16987][V][SSLClient.cpp:221] connect(): CA Certificate: Provided
[ 16993][V][SSLClient.cpp:222] connect(): Client Certificate: Provided
[ 17000][V][SSLClient.cpp:223] connect(): Private Key: Provided
[ 17005][V][ssl_client.cpp:238] start_ssl_client(): Free internal heap before TLS 342060
[ 17013][V][ssl_client.cpp:239] start_ssl_client(): Connecting to <anonymized>:8883
[ 17021][V][ssl_client.cpp:258] start_ssl_client(): Client pointer: 0x3ffc2520
[ 17382][V][ssl_client.cpp:267] start_ssl_client(): Seeding the random number generator
[ 17382][V][ssl_client.cpp:269] start_ssl_client(): Entropy context initialized
[ 17387][V][ssl_client.cpp:278] start_ssl_client(): Random number generator seeded, ret: 0
[ 17394][V][ssl_client.cpp:281] start_ssl_client(): Setting up the SSL/TLS defaults...
[ 17403][V][ssl_client.cpp:291] start_ssl_client(): SSL config defaults set, ret: 0
[ 17409][V][ssl_client.cpp:295] start_ssl_client(): Loading CA cert
[ 17421][V][ssl_client.cpp:354] start_ssl_client(): Loading CRT cert
[ 17426][V][ssl_client.cpp:362] start_ssl_client(): Loading private key
[ 17436][V][ssl_client.cpp:378] start_ssl_client(): Setting hostname for TLS session...
[ 17437][V][ssl_client.cpp:396] start_ssl_client(): Setting up IO callbacks...
[ 17442][V][ssl_client.cpp:400] start_ssl_client(): Setting timeout to 0
[ 17448][V][ssl_client.cpp:404] start_ssl_client(): Performing the SSL/TLS handshake...
[ 17456][V][ssl_client.cpp:431] start_ssl_client(): Protocol is SSLv3.0
[ 17462][E][ssl_client.cpp:438] start_ssl_client(): Failed to get Ciphersuite
[ 17469][E][ssl_client.cpp:447] start_ssl_client():  mbedtls_ssl_get_record_expansion returned -0xfffffffb
[ 17479][V][ssl_client.cpp:486] start_ssl_client(): Free internal heap after TLS 305768
[ 17486][V][SSLClient.cpp:232] connect(): Return value from start_ssl_client: 1
[ 17493][I][SSLClient.cpp:241] connect(): SSL connection established
[ 17499][V][ssl_client.cpp:577] send_ssl_data(): ssl_client->client: 0x3ffc2520
[ 17506][V][ssl_client.cpp:578] send_ssl_data(): ssl_client->handshake_timeout: 120000
[... 10 lines cut ...]

  #0  0x40083511:0x3ffb1eb0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008832d:0x3ffb1ed0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4008d6c9:0x3ffb1ef0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x400838aa:0x3ffb2020 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:360
      (inlined by) heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:345
  #4  0x40085365:0x3ffb2040 in esp_mbedtls_mem_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/port/esp_mem.c:46
  #5  0x400f6c72:0x3ffb2060 in mbedtls_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/platform.c:66
  #6  0x400e9660:0x3ffb2080 in mbedtls_ssl_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_tls.c:6914
  #7  0x400d56d9:0x3ffb20a0 in stop_ssl_socket(sslclient_context*, char const*, char const*, char const*) at .pio/libdeps/esp32dev/SSLClient/src/ssl_client.cpp:531
  #8  0x400d4a63:0x3ffb20e0 in SSLClient::stop() at .pio/libdeps/esp32dev/SSLClient/src/SSLClient.cpp:91
  #9  0x400d4bdd:0x3ffb2110 in SSLClient::write(unsigned char const*, unsigned int) at .pio/libdeps/esp32dev/SSLClient/src/SSLClient.cpp:375
      (inlined by) SSLClient::write(unsigned char const*, unsigned int) at .pio/libdeps/esp32dev/SSLClient/src/SSLClient.cpp:367
[... cut ...]
      (inlined by) PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*, bool) at .pio/libdeps/esp32dev/PubSubClient/src/PubSubClient.cpp:181
  #12 0x400d4612:0x3ffb2190 in PubSubClient::connect(char const*) at .pio/libdeps/esp32dev/PubSubClient/src/PubSubClient.cpp:166
  #13 0x400d258e:0x3ffb21c0 in reconnect() at src/main.cpp:84
  #14 0x400d3f4b:0x3ffb21e0 in loop() at src/main.cpp:251
  #15 0x400d8375:0x3ffb2290 in loopTask(void*) at /home/patrick/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: <anonymized>

Rebooting...
```

RobertByrnes, 2023-09-26:
[ 17462][E][ssl_client.cpp:438] start_ssl_client(): Failed to get Ciphersuite

der-elektrotechniker, 2023-09-26:
What does that mean? The server is set up correctly, I can connect with an MQTT-PC-Software without of problems...
Is the problem in the mbedtls library?

RobertByrnes, 2023-09-26:
**Point 1:**
```
const char* ssl_version = mbedtls_ssl_get_version(&ssl_client->ssl_ctx);
const char* ssl_ciphersuite = mbedtls_ssl_get_ciphersuite(&ssl_client->ssl_ctx);

if (ssl_version) {
    log_v("Protocol is %s", ssl_version);
} else {
    log_e("Failed to get Protocol");
}
if (ssl_ciphersuite) {
    log_v("Ciphersuite is %s", ssl_ciphersuite);
} else {
    log_e("Failed to get Ciphersuite");
}
```
This is the code you added and you now have the output: `[ 17462][E][ssl_client.cpp:438] start_ssl_client(): Failed to get Ciphersuite`.

**Point 2:**
Personally, I think this is the start of an issue. Why did it log that? Why didn't the call to mbedtls_ssl_get_ciphersuite return a string as expected? That's where I'd go looking. As I said before I start checking the contents of your ->ctx to make sure all is you would expect. Has the pointer to one of the certs become invalid due to scope?

**Point 3:**
[ 17387][V][ssl_client.cpp:278] start_ssl_client(): Random number generator seeded, ret: 0 - this line suggest RNG went OK yet when writing data it errored with:
[ 17521][E][ssl_client.cpp:45] _handle_error(): [send_ssl_data():589]: (-29696) SSL - No RNG was provided to the SSL module.

**Point 4:**
Possible Configuration Mismatch: This error could indicate a mismatch between the client's and server's supported cipher suites or SSL/TLS versions. - Might be worth double checking.

**Point 5:** 
Memory Issues: Lack of sufficient memory can sometimes lead to this issue. Ensure the ESP32  has enough heap memory available. Use functions like:

```
uint32_t freeHeap = ESP.getFreeHeap();
uint32_t totalHeap = ESP.getHeapSize();
uint32_t minFreeHeap = ESP.getMinFreeHeap();
UBaseType_t stackHWM = uxTaskGetStackHighWaterMark(NULL);
```

**Point 6:**
You clearly have an issue going on with memory:

[ 17559][V][ssl_client.cpp:530] stop_ssl_socket(): Freeing SSL context. Current ssl_ctx address: 0x3ffb9d14

assert failed: heap_caps_free heap_caps.c:360 (heap != NULL && "free() target pointer is outside heap areas")

These lines tell you that. So again I might try to add logging and a safety check by replacing these lines in stop_ssl_socket()

```
  // Free other SSL-related contexts and log their current addresses
  log_v("Freeing SSL context. Current ssl_ctx address: %p", (void *)&ssl_client->ssl_ctx);
  mbedtls_ssl_free(&ssl_client->ssl_ctx);

  log_v("Freeing SSL config. Current ssl_conf address: %p", (void *)&ssl_client->ssl_conf);
  mbedtls_ssl_config_free(&ssl_client->ssl_conf);

  log_v("Freeing DRBG context. Current drbg_ctx address: %p", (void *)&ssl_client->drbg_ctx);
  mbedtls_ctr_drbg_free(&ssl_client->drbg_ctx);

  log_v("Freeing entropy context. Current entropy_ctx address: %p", (void *)&ssl_client->entropy_ctx);
  mbedtls_entropy_free(&ssl_client->entropy_ctx);
```

with these lines....

```
mbedtls_ssl_free(&ssl_client->ssl_ctx);

if (&ssl_client->ssl_conf) {
    log_v("Freeing SSL config. Current ssl_conf address: %p", (void *)&ssl_client->ssl_conf);
    mbedtls_ssl_config_free(&ssl_client->ssl_conf);
} else {
    log_v("ssl_conf is invalid or not initialized.");
}

if (&ssl_client->drbg_ctx) {
    log_v("Freeing DRBG context. Current drbg_ctx address: %p", (void *)&ssl_client->drbg_ctx);
    mbedtls_ctr_drbg_free(&ssl_client->drbg_ctx);
} else {
    log_v("drbg_ctx is invalid or not initialized.");
}

if (&ssl_client->entropy_ctx) {
    log_v("Freeing entropy context. Current entropy_ctx address: %p", (void *)&ssl_client->entropy_ctx);
    mbedtls_entropy_free(&ssl_client->entropy_ctx);
} else {
    log_v("entropy_ctx is invalid or not initialized.");
}
```

der-elektrotechniker, 2023-09-27:
Hi @RobertByrnes ,

I solved the problem! The solution has two parts:

1. I removed the 3 years old `armmbed/mbedtls@^2.23.0` library. Since then (if I understand it right) the mbedtls Library of ESP32-framework itself is used, which is updated frequently. And since then the ciphersuite is showed correctly in the logs and the RNG issue is gone.
2. I realized, that I forgot to specify username and password for the mqtt server. After correcting this mistake a successfully connection was established.

So now it works perfectly!

Many thanks for your help!! And again: thank you so much for this awesome library! I hope you will keep it maintaining for a long long time!

RobertByrnes, 2023-09-28:
Glad you got it sorted. Will close this issue now.

closed 2023-09-28 by RobertByrnes