govorox/SSLClient #60: Not working in parallel with WiFiClientSecure

kymograph, 2023-10-29:
The lib is not working when used in parallel with the WiFiClientSecure lib. Both libs include ssl_client.h, but due to the changes in this lib, the function get_peer_fingerprint() cannot be found anymore, giving a compilation error.

Errors are probably depend on include order because of the two versions of ssl_client.h, so other errors might also lead to this issue.

RobertByrnes, 2023-12-05:
Hi, this may possibly get marked as a won't fix as I believe the lobrary was really made for use with insecure clients. Why were you trying to use along a secure client. This library was initially based on the WifiSecure lib you are using.

kymograph, 2023-12-05:
The use case is with devices, having two wireless interface, e.g., WiFi and GSM, such as the Liligo T-Call Modules  (https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800). In my use case, the WiFi interface is the preferred one, with fallback to the GSM interface. The WifiSecure lib can successfully be used on the WiFi interface. Using SSL on the SIM800 module however, is not straightforward. Thats where this lib comes in and I would love to use it on top of TinyGSM (https://github.com/vshymanskyy/TinyGSM) for the GSM connection path.

pontusmattiasdanielsson, 2024-02-15:
Having the same issue. Changing a few lines of code would fix it. Would be lovely to be able to use both libraries.

RobertByrnes, 2024-02-21:
@pontusmattiasdanielsson can you submit a PR for this?

live-alchemy, 2024-04-10:
@RobertByrnes  @pontusmattiasdanielsson adding a +1 here as our team has the same issue, we want to be able to fallback to WiFi if GSM is unavailable, but we need SSLClient to use GSM as our module SIM7600 doesn't support SSL natively.

RobertByrnes, 2024-04-10:
@live-alchemy Do you have any can share here such as a sketch or part of one to make it easier for me to take a look?

RobertByrnes, 2024-04-10:
https://github.com/ThrowTheSwitch/Unity/issues/722

RobertByrnes, 2024-04-10:
@kymograph @live-alchemy  @pontusmattiasdanielsson The latest version on PlatformIO / Arduino is 1.1.8 and should now support compilation with WifIClientSecure. Any further issues with this please open a new issue.

closed 2024-04-10 by RobertByrnes

reusables-official, 2024-04-10:
@RobertByrnes apologies for the novice question, is there a way to update this library manually if the platformio registry does not have the update yet? I went into the libdeps dir but these aren't git repos and I'm sure there's an easier way to do this. `pio pkg update` also didn't pick it up, perhaps I just have to wait for the update to propagate?

RobertByrnes, 2024-04-10:
@reusables-official no worries. Put in your platformio.ini file under the lib_deps section: digitaldragon/SSLClient@1.1.8

https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_deps.html

reusables-official, 2024-04-10:
@RobertByrnes yes! that was actually the first thing I tried, but I get this error back on save:
```
Resolving esp32-s3-devkitc-1 dependencies...
Library Manager: Installing digitaldragon/SSLClient @ 1.1.8
UnknownPackageError: Could not find the package with 'digitaldragon/SSLClient @ 1.1.8' requirements for your system 'darwin_x86_64'
```
I assume the issue isn't the last bit about system requirements, but maybe?

RobertByrnes, 2024-04-10:
@reusables-official sorry my bad! I tagged it as 1.1.8 but hadn't published yet. 

It is now published, and it should be there soon 😀 

In the meantime you can add: https://github.com/Govorox/SSLClient.git#1.1.8

RobertByrnes, 2024-04-10:
@reusables-official The other thing you can do is try 'pio system prune'

When the cache clears my system picked up the new package and installed it with the correct version

reusables-official, 2024-04-10:
@RobertByrnes sweet! got the latest version. it seems that some of the errors previously generated are indeed resolved, but there is one last one lingering! I think it's related to this package but honestly it's hard for me to tell, but I am seeing effects of your change in that `ssl__client` and not `ssl_client` is being imported:

```
/Users/enzo/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: 

.pio/build/esp32-s3-devkitc-1/lib0dc/libSSLClient.a(ssl__client.cpp.o):/Users/enzo/Documents/git/reusables/SRB-V3/.pio/libdeps/esp32-s3-devkitc-1/SSLClient/src/ssl__client.cpp:22: multiple definition of `pers';

.pio/build/esp32-s3-devkitc-1/libeb0/libWiFiClientSecure.a(ssl_client.cpp.o):/Users/enzo/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:27: first defined here
```

reusables-official, 2024-04-10:
@RobertByrnes I went into the source and changed the name of `pers` in `ssl__client.cpp` and it compiled fine! 😮‍💨
strange one, I guess there's another naming collision there. I could PR that if you think it's helpful.

RobertByrnes, 2024-04-10:
> @RobertByrnes I went into the source and changed the name of `pers` in `ssl__client.cpp` and it compiled fine! 😮‍💨
> strange one, I guess there's another naming collision there. I could PR that if you think it's helpful.

Yes please that would be helpful. That didn't come up in my env as it compiled.

Please can you include a snippet of your #includes 

Cheers

PR #69 (2024-04-10): fix: compatibility with WifiClientSecure.h
closes #60