govorox/SSLClient #62: Unable to make HTTPS POST with client certificate

cezarg1410, 2023-12-04:
Hello,
I have really strange issue when trying to make HTTP POST with SSL from ESP32 with SIM7600G.
The thing is it works from time to time. Probably 1-2 out of 10 calls are successfull. For the rest the library returns -2 or -3 status code.
Here is my full code:

```
#include <SSLClient.h>
#include "certs.h"
#include <HardwareSerial.h>

#define GSM_RX 27
#define GSM_TX 14

#define TEENSY_RX 16
#define TEENSY_TX 17

#define MODEM_PWRKEY 25

// Select your modem:
#define TINY_GSM_MODEM_SIM7600
#define TINY_GSM_RX_BUFFER 1024

#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>

HardwareSerial SerialAT(1);      //gsm
HardwareSerial SerialTeensy(2);  //teensy

//============================================
#define TINY_GSM_DEBUG SerialMon

#ifdef TINY_GSM_DEBUG
#define SerialMon Serial
#endif

#define TINY_GSM_USE_GPRS true
//============================================
//#define DUMP_AT_COMMANDS
#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif
//============================================

TinyGsmClient gsm_transpor_layer(modem, 0);
SSLClient secure_presentation_layer(&gsm_transpor_layer);
HttpClient client = HttpClient(secure_presentation_layer, server, 443);

unsigned long prevFinished = 0;

void setup() {
#ifdef TINY_GSM_DEBUG
  SerialMon.begin(115200);
  delay(10);
#endif

  logTextLine("Wait...");

  SerialAT.begin(115200, SERIAL_8N1, GSM_RX, GSM_TX);
  while (!SerialAT)
    ;

  SerialTeensy.begin(115200, SERIAL_8N1, TEENSY_RX, TEENSY_TX);
  while (!SerialTeensy)
    ;

  setupModem();
[... 273 lines cut ...]
[539796][E][SSLClient.cpp:235] connect(): start_ssl_client failed: 0
[539802][D][SSLClient.cpp:90] stop(): Stopping ssl client
[539807][D][ssl_client.cpp:791] stop_ssl_socket(): Cleaning SSL connection.
[539814][D][ssl_client.cpp:795] stop_ssl_socket(): Stopping SSL client. Current client pointer address: 0x3ffc2468
[539885][D][ssl_client.cpp:816] stop_ssl_socket(): Freeing SSL context. Current ssl_ctx address: 0x3ffba7d8
[539885][D][ssl_client.cpp:819] stop_ssl_socket(): Freeing SSL config. Current ssl_conf address: 0x3ffba9f8
[539893][D][ssl_client.cpp:822] stop_ssl_socket(): Freeing DRBG context. Current drbg_ctx address: 0x3ffbaae0
[539903][D][ssl_client.cpp:825] stop_ssl_socket(): Freeing entropy context. Current entropy_ctx address: 0x3ffbab30
[539913][D][ssl_client.cpp:828] stop_ssl_socket(): Finished cleaning SSL connection.
[539920][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539926][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539932][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539938][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539944][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539950][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539956][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539962][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539968][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539974][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539980][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539986][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539992][W][SSLClient.cpp:369] write(): SSLClient is not connected.
[539998][W][SSLClient.cpp:369] write(): SSLClient is not connected.
Status code: -3
```

RobertByrnes, 2023-12-05:
Hi, the -3 looks like a timeout coming from the ArduinoHttpClient. Also, it looks due to the ssl handshake failing.

I have had this before and fixed when using ArduinoHttpClient the call to client.connectionKeepAlive() before startRequest is called. Although technically this shouldn't be necessary.

Also, check your ArduinoHttpClient and SSLClient timeout values and play at adjusting them to longer. Maybe go higher and work back.

Your post request structure for using the ArduinoHttpClient looks good.

With TinyGsm is tend to log out a connection strength operator periodically as well to help diagnose connection issues.

RobertByrnes, 2023-12-07:
Hi, @ilgarbenli I have used this lib with SIM800L and ArduinoHttpClient successfully.

Just using client.post() method (look at ArduinoHttpClient examples.

Using client.write() only

And also (my favourite method for completeness, structure and clarity) using a complex and explicit use of ArduinoHttpClient - one of their e ample shows this. Pretty sure you are better off in this order.

1. client.connect
2. client.connectionKeepAlive
3. client.startRequest(endpoint, method)
4. client.sendHeader() content type
5. client.sendHeader() content len
6. client.

ilgarbenli, 2023-12-09:
@RobertByrnes thank you for your respond. I tried several things and couldn't make it unfortunately. I guess I'm doing something wrong with my code. I'll open new issue of mine with my full of code.  Would you review it when you have a chance

ilgarbenli, 2023-12-09:
@cezarg1410 what is resource in client.post(resource); ?

RobertByrnes, 2023-12-09:
Resource = "/something"

As in https://api.example.com/something

You have already given the host when you connected in the connect method as "api.example.com" so now the endpoint or resource is "/something"

ilgarbenli, 2023-12-10:
> Resource = "/something"
> 
> As in https://api.example.com/something
> 
> You have already given the host when you connected in the connect method as "api.example.com" so now the endpoint or resource is "/something"

Thank you for your respond and thank you for your help. My code is working now.

cezarg1410, 2023-12-14:
Unfortunately - no luck for me.
I tried option with connectionKeepAlive() but it does not change anything.
SSL Handshake should be fine - i am able to POST identical request from postman without issues.

Looks like, the library or the sim7600 is not able to handle the client certificate - when i remove those two lines:
```
  secure_presentation_layer.setCertificate(client_crt);
  secure_presentation_layer.setPrivateKey(client_key);
```
then it works fine every time (of course i am getting then HTTP 403 response because my server requires proper client certificate).

I updated all libraries (TinyGSM, SSLClient and ArduinoHttpClient) to the newest versions and something has changed. Now i am getting -3 error every single time : D

I am starting to loose hope :) Looks like i should resign from authenticating client with its certificate, but obviously i would rather not to do it - it is in my opinion the best and the most secure method for exposing API to the internet.

ilgarbenli, 2023-12-14:
@cezarg1410 may it works for your case? https://github.com/Samorange1/Connect-ESP32-to-Firebase-via-GSM
it bypass ssl connecction with a server

cezarg1410, 2023-12-14:
Thanks for that! Looks interesting, will get deep dive into it

ilgarbenli, 2023-12-14:
> Thanks for that! Looks interesting, will get deep dive into it

please keep contact together, It worked for me. But also I'm struggling about 403 forbidden issue for another client. Maybe we can help together

cezarg1410, 2024-01-11:
No luck for me.
I decided to remove client certificate authentication from my solution and instead i am using API token authentication on my rest API with HTTPS.
It works just fine and i stopped wasting time trying to make it working.

RobertByrnes, 2024-05-01:
solved by v1.2.0

closed 2024-05-01 by RobertByrnes