govorox/SSLClient #64: HTTPS POST request error with SSL/TLS. Can't connect endpoint.

ilgarbenli, 2023-12-09:
Hello 
I have trouble about making HTTPS POST request with SSL. With using LilyGo T-Call SIM800L_IP5306_VERSION_20190610. 
I'm trying to use google geolocation api and it only allows https secure connections, with JSON format requests. ([api link](https://developers.google.com/maps/documentation/geolocation/overview?hl=en))

I feel I'm doing something wrong in code. Could someone help about this.

its code I'm using. (It based on examples in this library.)

```
/**************************************************************
 *
 * ESP32 LilyGo-T-Call-SIM800 Example
 *
 * HTTPS (TLS/SLL) with CA Certificate via "TinyGsm.h": https://github.com/vshymanskyy/TinyGSM
 * Tested on SIM800L_IP5306_VERSION_20190610 (v1.3) (R14.18)
 *
 * About board: https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800
 * About SIM800L_IP5306 v1.3: https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/doc/SIM800L_IP5306.MD    
 * Base example: https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_TinyGSM  
 * 
 **************************************************************/

#include <Arduino.h>
#if defined ARDUINO_ARCH_ESP8266
#include <ESP8266WiFi.h>
#elif defined ARDUINO_ARCH_ESP32
#include <WiFi.h>
#else
#error Wrong platform
#endif

#define DUMP_AT_COMMANDS

#include <Wire.h>
#include "SSLClient.h"
//To make http request esay: https://github.com/arduino-libraries/ArduinoHttpClient
#include <ArduinoHttpClient.h>

//Please enter your CA certificate in ca_cert.h
#include "ca_cert.h"

// ESP32 LilyGo-T-Call-SIM800 SIM800L_IP5306_VERSION_20190610 (v1.3) pins definition
#define MODEM_UART_BAUD 9600
#define MODEM_RST 5
#define MODEM_PWRKEY 4
#define MODEM_POWER_ON 23
#define MODEM_TX 27
#define MODEM_RX 26
#define I2C_SDA 21
#define I2C_SCL 22
#define LED_PIN 13
#define IP5306_ADDR 0x75
#define IP5306_REG_SYS_CTL0 0x00

// Set serial for debug console (to the Serial Monitor)
#define SerialMon Serial
// Set serial for AT commands (to the SIM800 module)
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800   // Modem is SIM800
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb

#define MAX_CONNECTION_TIMEOUT 5000
#define MAX_WIFI_SCAN 127

// Include after TinyGSM definitions
#include <TinyGsmClient.h>

// Your GPRS credentials (leave empty, if missing)
const char apn[] = "internet";       // Your APN
const char gprs_user[] = ""; // User
const char gprs_pass[] = ""; // Password
const char simPIN[] = "";    // SIM card PIN code, if any

const char hostname[] = "https://www.googleapis.com";
String resource = "/geolocation/v1/geolocate?key=AIzaSyB-JNrD3wkw8uRnYShamLkb6LJ1pziJMCA";
int port = 443;

// Layers stack
TinyGsm sim_modem(SerialAT);
TinyGsmClientSecure gsm_transpor_layer(sim_modem);
SSLClient secure_presentation_layer(&gsm_transpor_layer);
HttpClient http_client = HttpClient(secure_presentation_layer, hostname, port);

// Power configuration for SIM800L_IP5306_VERSION_20190610 (v1.3) board
bool setupPMU()
{
  bool en = true;
  Wire.begin(I2C_SDA, I2C_SCL);
  Wire.beginTransmission(IP5306_ADDR);
  Wire.write(IP5306_REG_SYS_CTL0);
  if (en)
  {
    Wire.write(0x37);
  }
  else
  {
    Wire.write(0x35);
  }
  return Wire.endTransmission() == 0;
}

// Modem initial setup (cold start)
void setupModem()
{
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_PWRKEY, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  pinMode(LED_PIN, OUTPUT);
  
  // Reset pin high
  digitalWrite(MODEM_RST, HIGH);

  // Turn on the Modem power first
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Pull down PWRKEY for more than 1 second according to manual requirements
  digitalWrite(MODEM_PWRKEY, HIGH);
  delay(200);
  digitalWrite(MODEM_PWRKEY, LOW);
  delay(1200);
  digitalWrite(MODEM_PWRKEY, HIGH);

  // Initialize the indicator as an output
  digitalWrite(LED_PIN, LOW);
}

void setup()
{
  SerialMon.begin(9600);
  delay(100);

  // Start board power management
  if (!setupPMU())
  {
    Serial.println("Setting board power management error");
  }

  // Set SIM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);

  //Add CA Certificate
  secure_presentation_layer.setCACert(root_ca);

  // SIM modem initial setup
  setupModem();
}

void loop()
{
  SerialMon.print("Initializing modem...");
  if (!sim_modem.init())
  {
    SerialMon.print(" fail... restarting modem...");
    setupModem();
    // Restart takes quite some time
    // Use modem.init() if you don't need the complete restart
    if (!sim_modem.restart())
    {
      SerialMon.println(" fail... even after restart");
      return;
    }
  }
  SerialMon.println(" OK");

  // General information
  String name = sim_modem.getModemName();
  Serial.println("Modem Name: " + name);
  String modem_info = sim_modem.getModemInfo();
  Serial.println("Modem Info: " + modem_info);

  // Unlock your SIM card with a PIN if needed
  if (strlen(simPIN) && sim_modem.getSimStatus() != 3)
  {
    sim_modem.simUnlock(simPIN);
  }

  // Wait for network availability
  SerialMon.print("Waiting for network...");
  if (!sim_modem.waitForNetwork(240000L))
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");
  
  // Connect to the GPRS network
  SerialMon.print("Connecting to network...");
  if (!sim_modem.isNetworkConnected())
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  // Connect to APN
  SerialMon.print(F("Connecting to APN: "));
  SerialMon.print(apn);
  if (!sim_modem.gprsConnect(apn, gprs_user, gprs_pass))
  {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  digitalWrite(LED_PIN, HIGH);
  SerialMon.println(" OK");

  // More info..
  Serial.println("");
  String ccid = sim_modem.getSimCCID();
  Serial.println("CCID: " + ccid);
  String imei = sim_modem.getIMEI();
  Serial.println("IMEI: " + imei);
  String cop = sim_modem.getOperator();
  Serial.println("Operator: " + cop);
  IPAddress local = sim_modem.localIP();
  Serial.println("Local IP: " + String(local));
  int csq = sim_modem.getSignalQuality();
  Serial.println("Signal quality: " + String(csq));

  /// HTTP Test
  if (sim_modem.isGprsConnected())
  {
  Serial.println("");
   

    if (http_client.connect(hostname, port)) {
        Serial.println("Connected to API endpoint\n");
    } else {
         Serial.println("HTTPS error\n");
    }

  Serial.println("Performing HTTPS POST request with TLS/SSL");
  String body = "{\"considerIP\":false,\"wifiAccessPoints\":" + getSurroundingWiFiJson() + "}";
  http_client.beginRequest();
  http_client.post(resource);
  http_client.sendHeader("User-Agent", "ESP32");
  http_client.sendHeader("Content-Type", "application/json");
  http_client.sendHeader("Content-Length", String(body.length()));
  http_client.connectionKeepAlive();
  http_client.beginBody();
  http_client.println(body);
  http_client.endRequest();

    int status_code = http_client.responseStatusCode();
    String response = http_client.responseBody();

    Serial.print("Status code: ");
    Serial.println(status_code);
    Serial.print("Response: ");
    Serial.println(response);
    Serial.println("----------: ");
    Serial.println(body);
    http_client.stop();

  }
  else
  {
    Serial.println("...not connected");
  }

  // Disconnect GPRS
  sim_modem.gprsDisconnect();
  SerialMon.println("GPRS disconnected");
  digitalWrite(LED_PIN, LOW);

  //Turn off the moden (if use, you need run setupModem() again)
  //sim_modem.poweroff();
  //SerialMon.println("Modem poweroff");
  //delay(1000);
  //setupModem();

  delay(15000);
}

String getSurroundingWiFiJson()
{
    String wifiArray = "[";

    int8_t numWifi = WiFi.scanNetworks();
    if (numWifi > MAX_WIFI_SCAN)
    {
        numWifi = MAX_WIFI_SCAN;
    }

    for (uint8_t i = 0; i < numWifi; i++)
    {
        wifiArray += "{\"macAddress\":\"" + WiFi.BSSIDstr(i) + "\",";
        wifiArray += "\"signalStrength\":" + String(WiFi.RSSI(i)) + ",";
        wifiArray += "\"channel\":" + String(WiFi.channel(i)) + "}";
        if (i < (numWifi - 1))
        {
            wifiArray += ",\n";
        }
    }
    WiFi.scanDelete();
    wifiArray += "]";
    return wifiArray;
}
```

And also I tried several ca_certs with GET request for making connection test, it only worked with the certificate in the example that has already expired. (It confuses too me)

[image]

I am open to any suggestion or anything that can help.

ilgarbenli, 2023-12-09:
I used String getSurroundingWiFiJson() function from https://github.com/gmag11/WifiLocation/ so maybe also @gmag11 may help

ilgarbenli, 2023-12-09:
do you have any idea why I can't connect googleapis @RobertByrnes

gmag11, 2023-12-09:
Hi, you can use this:

```
openssl s_client -showcerts -connect www.googleapis.com:443
```
Response is:

```
CONNECTED(00000003)
depth=2 C = US, O = Google Trust Services LLC, CN = GTS Root R1
verify return:1
depth=1 C = US, O = Google Trust Services LLC, CN = GTS CA 1C3
verify return:1
depth=0 CN = upload.video.google.com
verify return:1
---
Certificate chain
 0 s:CN = upload.video.google.com
   i:C = US, O = Google Trust Services LLC, CN = GTS CA 1C3
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: RSA-SHA256
   v:NotBefore: Nov 20 08:08:50 2023 GMT; NotAfter: Feb 12 08:08:49 2024 GMT
-----BEGIN CERTIFICATE-----
MIIGEjCCBPqgAwIBAgIQDW4VistptnES0TZTlfkMYzANBgkqhkiG9w0BAQsFADBG
MQswCQYDVQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExM
QzETMBEGA1UEAxMKR1RTIENBIDFDMzAeFw0yMzExMjAwODA4NTBaFw0yNDAyMTIw
ODA4NDlaMCIxIDAeBgNVBAMTF3VwbG9hZC52aWRlby5nb29nbGUuY29tMFkwEwYH
KoZIzj0CAQYIKoZIzj0DAQcDQgAEVV8eiSunJ1y3il3cZM/ryPIj0pE+LewQYTkv
1kw/AJ4UGY9JGX76hLfWxWOUs+hEZil51ZpGQLowyPwH+qFuPqOCA+kwggPlMA4G
A1UdDwEB/wQEAwIHgDATBgNVHSUEDDAKBggrBgEFBQcDATAMBgNVHRMBAf8EAjAA
MB0GA1UdDgQWBBRLtbRGlCVTjb0hXaDTXU5QtPUeUjAfBgNVHSMEGDAWgBSKdH+v
hc3ulc09nNDiRhTzcTUdJzBqBggrBgEFBQcBAQReMFwwJwYIKwYBBQUHMAGGG2h0
dHA6Ly9vY3NwLnBraS5nb29nL2d0czFjMzAxBggrBgEFBQcwAoYlaHR0cDovL3Br
aS5nb29nL3JlcG8vY2VydHMvZ3RzMWMzLmRlcjCCAZgGA1UdEQSCAY8wggGLghd1
cGxvYWQudmlkZW8uZ29vZ2xlLmNvbYIUKi5jbGllbnRzLmdvb2dsZS5jb22CESou
ZG9jcy5nb29nbGUuY29tghIqLmRyaXZlLmdvb2dsZS5jb22CEyouZ2RhdGEueW91
dHViZS5jb22CECouZ29vZ2xlYXBpcy5jb22CEyoucGhvdG9zLmdvb2dsZS5jb22C
FyoueW91dHViZS0zcmQtcGFydHkuY29tghF1cGxvYWQuZ29vZ2xlLmNvbYITKi51
cGxvYWQuZ29vZ2xlLmNvbYISdXBsb2FkLnlvdXR1YmUuY29tghQqLnVwbG9hZC55
b3V0dWJlLmNvbYIfdXBsb2Fkcy5zdGFnZS5nZGF0YS55b3V0dWJlLmNvbYIVYmct
Y2FsbC1kb25hdGlvbi5nb29nghtiZy1jYWxsLWRvbmF0aW9uLWFscGhhLmdvb2eC
HGJnLWNhbGwtZG9uYXRpb24tY2FuYXJ5Lmdvb2eCGWJnLWNhbGwtZG9uYXRpb24t
ZGV2Lmdvb2cwIQYDVR0gBBowGDAIBgZngQwBAgEwDAYKKwYBBAHWeQIFAzA8BgNV
HR8ENTAzMDGgL6AthitodHRwOi8vY3Jscy5wa2kuZ29vZy9ndHMxYzMvemRBVHQw
RXhfRmsuY3JsMIIBBQYKKwYBBAHWeQIEAgSB9gSB8wDxAHYAO1N3dT4tuYBOizBb
Bv5AO2fYT8P0x70ADS1yb+H61BcAAAGL6/1dIwAABAMARzBFAiAtPr+kGyD/r1OL
fLRHzG9CBJK9X84wLvyahFObiY3MfQIhAODHguLSKrsKJi89jPdS+QXFzf9W2ZD1
d5wfPG69FoSSAHcAdv+IPwq2+5VRwmHM9Ye6NLSkzbsp3GhCCp/mZ0xaOnQAAAGL
6/1ZWwAABAMASDBGAiEA1CiIrFNPniaoXJ0Nk5zYz0ZmiGvFc4snw8TyrRnduX8C
IQDewMD1uni5Ux32bRrYqTVsNM/KA7wr9ZO8OZDyfCWV0DANBgkqhkiG9w0BAQsF
AAOCAQEA0g9fPApzdqcxH/tmRY8Am4b748Mt14K1MMWMCu5/CLcVDoou00fjiFGW
/R8xzaimZ9GtNzdOpMrMJd1bedROh7cweVbp1095L4hEOvLlqaIqEcfOTqT1csyQ
vZWCIGVc2gFOlLACJ6l2866sD77sWtf29nty40VF80mUryTxKqQv6FhY4jEwRq9F
tMP80XUrXgeMAwjZ0HyvHE7Z1yThExwme7SocVXFdDO/qCvmM18zLgnaDfETybt3
uI1InseIQ9l2DERqlCX1s4hTrd3iTGNOEkhgMtfRytdEOmZWJXpmSSQxvfabZUlU
pdEptrqDy1vgw4xsZcCCJyGwMI+LgQ==
-----END CERTIFICATE-----
 1 s:C = US, O = Google Trust Services LLC, CN = GTS CA 1C3
   i:C = US, O = Google Trust Services LLC, CN = GTS Root R1
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Aug 13 00:00:42 2020 GMT; NotAfter: Sep 30 00:00:42 2027 GMT
-----BEGIN CERTIFICATE-----
MIIFljCCA36gAwIBAgINAgO8U1lrNMcY9QFQZjANBgkqhkiG9w0BAQsFADBHMQsw
CQYDVQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExMQzEU
MBIGA1UEAxMLR1RTIFJvb3QgUjEwHhcNMjAwODEzMDAwMDQyWhcNMjcwOTMwMDAw
MDQyWjBGMQswCQYDVQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZp
Y2VzIExMQzETMBEGA1UEAxMKR1RTIENBIDFDMzCCASIwDQYJKoZIhvcNAQEBBQAD
ggEPADCCAQoCggEBAPWI3+dijB43+DdCkH9sh9D7ZYIl/ejLa6T/belaI+KZ9hzp
kgOZE3wJCor6QtZeViSqejOEH9Hpabu5dOxXTGZok3c3VVP+ORBNtzS7XyV3NzsX
lOo85Z3VvMO0Q+sup0fvsEQRY9i0QYXdQTBIkxu/t/bgRQIh4JZCF8/ZK2VWNAcm
BA2o/X3KLu/qSHw3TT8An4Pf73WELnlXXPxXbhqW//yMmqaZviXZf5YsBvcRKgKA
[... 56 lines cut ...]
-----END CERTIFICATE-----
---
Server certificate
subject=CN = upload.video.google.com
issuer=C = US, O = Google Trust Services LLC, CN = GTS CA 1C3
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 4690 bytes and written 400 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
```
Root CA is the latest certificate. You can copy and paste it in your code

ilgarbenli, 2023-12-10:
@gmag11 second one is worked for me, do I need other certs after and what are they others?

And thank you for your respond, thank you for your help. It worked.

closed 2023-12-10 by ilgarbenli