govorox/SSLClient #66: Connecting 2 Client Conflict

ilgarbenli, 2023-12-13:
Hello everyone, 

I'm trying to getting datas from a client and then sending data to another client. The probem is, one client needs ssl/tls connection (which is no problem I can connect and take datas) and other client is using http port:80 (probllem is here). 

When I tried to connect and send data individually to this http server its okay everythings work. But when I try with other ssl/tls cleint I think there is some kind of conflict. 

this is my structure of clients.

```
// Layers stack
TinyGsm sim_modem(SerialAT);
TinyGsmClient gsm_transpor_layer(sim_modem);
SSLClient secure_presentation_layer(&gsm_transpor_layer);
HttpClient http_client = HttpClient(secure_presentation_layer, geolocation_hostname, geolocation_port);
HttpClient    database_client(gsm_transpor_layer, database_server, database_port);
```

and the POST response from database_client:
```
status code: 403
Response: <!DOCTYPE html>
<html style="height:100%">
<head>
<meta name="viewport" content="width=device-width, initial-scale=1, shrink-to-fit=no" />
<title> 403 Forbidden
</title></head>
<body style="color: #444; margin:0;font: normal 14px/20px Arial, Helvetica, sans-serif; height:100%; background-color: #fff;">
<div style="height:auto; min-height:100%; ">     <div style="text-align: center; width:800px; margin-left: -400px; position:absolute; top: 30%; left:50%;">
        <h1 style="margin:0; font-size:150px; line-height:150px; font-weight:bold;">403</h1>
<h2 style="margin-top:20px;font-size: 30px;">Forbidden
</h2>
<p>Access to this resource on the server is denied!</p>
</div></div></body></html>
```
I guess its no problem with server I can send data induvidually, I also tested with https://reqbin.com. there is no problem.

But when I use with other client, its not working.

I am open to any suggestion.

ilgarbenli, 2023-12-13:
I realized problem is not about 2 client. Problem is about client. When I try to POST json request to my server it returns this 403 error, but not reqbin.com gives any error, even it works perfect. 
Do you have any idea @govorox @RobertByrnes

RobertByrnes, 2023-12-13:
2 things I'd check. You are using the same gsm layer for both clients. This could make state tracking cery tricky for application. First of make the 2 stacks have a unique set of objects for simplicity, safety, and clarity. Also, 403 forbidden, yes is client error but also check the path you are trying to access is correct and is available to your client from the web servers side ....

ilgarbenli, 2023-12-13:
@RobertByrnes first of all thank you for your respond. 
I made a test with this repo : https://github.com/Samorange1/Connect-ESP32-to-Firebase-via-GSM and it worked. I can send data to firebase with this method. But when I change code to sending json request to my server it gives 403 error, each time even if connects endpoint. I dont know how to make things works but I'm trying to send json data.
Actually my last php code works well with [reqbin.com](https://reqbin.com/) but returns 403 with sim800l request. 

Its like something refuses arduino request

RobertByrnes, 2023-12-13:
Another thing I've done before is to build the request in Postman. Then you have visibility over exactly which headers you are sending and can view the full requests and responses in the console. Can help

ilgarbenli, 2023-12-14:
@RobertByrnes I'm just trying Postman, its also working with postman requests, and not suspicious headers there. I changed body types to x-www-form-urlencoded in reqbin, it returns same 403 forbidden, switching back application/json made things work again.

I confused about is my headers wrong or another permission problem exist. It's not make sense for me.

this is my last structure:

```
#include <Arduino.h>
#if defined ARDUINO_ARCH_ESP8266
#include <ESP8266WiFi.h>
#elif defined ARDUINO_ARCH_ESP32
#include <WiFi.h>
#else
#error Wrong platform
#endif

#include <Wire.h>
#include "SSLClient.h"
//To make http request esay: https://github.com/arduino-libraries/ArduinoHttpClient
#include <ArduinoHttpClient.h>

//Please enter your CA certificate in ca_cert.h
#include "ca_cert.h"

// ESP32 LilyGo-T-Call-SIM800 SIM800L_IP5306_VERSION_20190610 (v1.3) pins definition
#define MODEM_UART_BAUD 9600
#define MODEM_RST 5
#define MODEM_PWRKEY 4
#define MODEM_POWER_ON 23
#define MODEM_TX 27
#define MODEM_RX 26
#define I2C_SDA 21
#define I2C_SCL 22
#define LED_PIN 13
#define IP5306_ADDR 0x75
#define IP5306_REG_SYS_CTL0 0x00

// Set serial for debug console (to the Serial Monitor)
#define SerialMon Serial
// Set serial for AT commands (to the SIM800 module)
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800   // Modem is SIM800
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb

#define MAX_CONNECTION_TIMEOUT 5000
#define MAX_WIFI_SCAN 127

// Include after TinyGSM definitions
#include <TinyGsmClient.h>

// Your GPRS credentials (leave empty, if missing)
const char apn[] = "internet";       // Your APN
const char gprs_user[] = ""; // User
const char gprs_pass[] = ""; // Password
const char simPIN[] = "";    // SIM card PIN code, if any

const char hostname[] = "******.com";
const String resource = "/gpt.php";
int port = 443;

// Layers stack
TinyGsm sim_modem(SerialAT);
TinyGsmClient gsm_transpor_layer(sim_modem);
SSLClient secure_presentation_layer(&gsm_transpor_layer);
HttpClient http_client = HttpClient(secure_presentation_layer, hostname, port);

// Power configuration for SIM800L_IP5306_VERSION_20190610 (v1.3) board
bool setupPMU()
[... 252 lines cut ...]
$response = file_get_contents("https://" . $geolocation_hostname . $geolocation_resource, false, $context);

// Google Geolocation API'den gelen cevabı decode et
$location_data = json_decode($response, true);

// Firebase'e gönderilecek verileri hazırla
$latitude = $location_data['location']['lat'];
$longitude = $location_data['location']['lng'];

$_devicestatus = array(
    'latitude' => $latitude,
    'longitude' => $longitude
);

// Firebase'e veriyi gönder
$firebase = new \Firebase\FirebaseLib(DEFAULT_URL, DEFAULT_TOKEN);
$firebase->update($DEFAULT_PATH, $_devicestatus);

// Başarı mesajını döndür
echo "POST_SUCCESSFUL";

?>
```

may it about file_get_contents() func?

ilgarbenli, 2023-12-14:
and also in postman, changing json to xml mades return 403

ilgarbenli, 2023-12-14:
its my  arduino side struct 
[image]

and this is reqbin request (its working)
[image]

do you see any mistake?

RobertByrnes, 2023-12-14:
The only thing that jumps out at me is thay you have no space after the : in your Content-Type and Content-Length headers...

ilgarbenli, 2023-12-14:
@RobertByrnes Didn't worked unfortunatelly. 

But its my own vds server, do you have any advice to change some settings about security or permission?

ilgarbenli, 2023-12-16:
@RobertByrnes I found what causes 403 error, its line which includes file_get_contents() func in my php code. I changed this and it worked.

RobertByrnes, 2023-12-16:
@ilgarbenli glad to hear you managed to resolve the issue

closed 2023-12-29 by ilgarbenli