govorox/SSLClient #67: does SSLClient support SNI - Server Name Identification - ?

kgmuzungu, 2024-02-15:
Some MQTT broker require SNI support for authentication (e.g. see here https://community.hivemq.com/t/connecting-to-private-broker-with-esp32-sim800l/2126) Does SSLClient support that? How could I set this field?

WebDust21, 2024-02-18:
> Does SSLClient support that?

It is already part of the connection process, based on the hostname you provide for connection.  The SNI is part of the connection sequence here:
https://github.com/govorox/SSLClient/blob/eee87188d2a63f115681c836083d4cbc9f45e76d/src/ssl_client.cpp#L324

with the actual routine here:
https://github.com/govorox/SSLClient/blob/eee87188d2a63f115681c836083d4cbc9f45e76d/src/ssl_client.cpp#L599

> Do I have to enable SNI on the modem side (in my case a SIM800L)
No, as this is not utilizing any of the SSL/TLS functionality in your modem.  The entire SSL functionality is completely handled in the main processor.

RobertByrnes, 2024-03-08:
@kgmuzungu are you good with this now? If you have had success I will mark this issue as closed. If not please update the issue with more about the exact problem you are facing. Thank you.

RobertByrnes, 2024-04-30:
Closing as no more messages. Closing with the understanding that yes this library supports server name Identification.

closed 2024-04-30 by RobertByrnes