govorox/SSLClient #68: UNKNOWN ERROR CODE (001D)

githubnu9, 2024-03-04:
Hi all,

I'm trying to connect with my own mqtt broker via a gsm800L while using the example ```mqtt_gsm_sim800l_azure_x509_device_Twin.ino``` with a self-generated CA/Client/Private key.

But im receiving ```error code (001D)```

```
[  9390][V][ssl_client.cpp:135] client_net_recv_timeout(): SSL client RX (received=5 expected=5 in 0ms)
[  9399][V][ssl_client.cpp:118] client_net_recv_timeout(): Timeout set to 0
[  9406][V][ssl_client.cpp:135] client_net_recv_timeout(): SSL client RX (received=40 expected=40 in 0ms)
[  9416][V][ssl_client.cpp:722] perform_ssl_handshake(): Protocol is TLSv1.2 Ciphersuite is TLS-ECDHE-RSA-WITH-AES-256-GCM-SHA384
[  9427][E][ssl_client.cpp:728] perform_ssl_handshake(): mbedtls_ssl_get_record_expansion returned -0xffffffe3
[  9437][D][ssl_client.cpp:791] stop_ssl_socket(): Cleaning SSL connection.
[  9443][D][ssl_client.cpp:795] stop_ssl_socket(): Stopping SSL client. Current client pointer address: 0x3ffc1e04
[  9467][D][ssl_client.cpp:800] stop_ssl_socket(): Freeing CA cert. Current ca_cert address: 0x3ffba2e4
[  9467][D][ssl_client.cpp:808] stop_ssl_socket(): Freeing client cert and client key. Current client_cert address: 0x3ffba43c, client_key address: 0x3ffba594
[  9479][D][ssl_client.cpp:816] stop_ssl_socket(): Freeing SSL context. Current ssl_ctx address: 0x3ffb9d14
[  9488][D][ssl_client.cpp:819] stop_ssl_socket(): Freeing SSL config. Current ssl_conf address: 0x3ffb9f34
[  9498][D][ssl_client.cpp:822] stop_ssl_socket(): Freeing DRBG context. Current drbg_ctx address: 0x3ffba01c
[  9507][D][ssl_client.cpp:825] stop_ssl_socket(): Freeing entropy context. Current entropy_ctx address: 0x3ffba06c
[  9517][D][ssl_client.cpp:828] stop_ssl_socket(): Finished cleaning SSL connection.
[  9525][D][ssl_client.cpp:249] cleanup(): Free internal heap after TLS 303284
[  9532][E][ssl_client.cpp:45] _handle_error(): [start_ssl_client():353]: (29) UNKNOWN ERROR CODE (001D)
[  9541][V][SSLClient.cpp:232] connect(): Return value from start_ssl_client: 0
[  9548][E][SSLClient.cpp:235] connect(): start_ssl_client failed: 0
[  9554][D][SSLClient.cpp:90] stop(): Stopping ssl client
[  9559][D][ssl_client.cpp:791] stop_ssl_socket(): Cleaning SSL connection.
[  9566][D][ssl_client.cpp:795] stop_ssl_socket(): Stopping SSL client. Current client pointer address: 0x3ffc1e04
[  9584][D][ssl_client.cpp:816] stop_ssl_socket(): Freeing SSL context. Current ssl_ctx address: 0x3ffb9d14
[  9585][D][ssl_client.cpp:819] stop_ssl_socket(): Freeing SSL config. Current ssl_conf address: 0x3ffb9f34
[  9595][D][ssl_client.cpp:822] stop_ssl_socket(): Freeing DRBG context. Current drbg_ctx address: 0x3ffba01c
[  9604][D][ssl_client.cpp:825] stop_ssl_socket(): Freeing entropy context. Current entropy_ctx address: 0x3ffba06c
[  9615][D][ssl_client.cpp:828] stop_ssl_socket(): Finished cleaning SSL connection.
```

In Mosquitto i see the following logging:
```
1709589226: New connection from 62.xxx.xxx.xxx:14080 on port 8883.
1709589230: Client <unknown> closed its connection.
```

I've checked the certificates and private keys. Those work while using ```WiFiClientSecure.h``` via WiFi with the same certificate and keys. Does anyone know where ```ERROR 001D``` comes from?

any help appreciated

RobertByrnes, 2024-04-29:
@githubnu9 https://github.com/govorox/SSLClient/pull/78 Please see this PR and the comments in issue https://github.com/govorox/SSLClient/issues/71 This may now be fixed

RobertByrnes, 2024-05-01:
Closing as solved by v1.2.0

closed 2024-05-01 by RobertByrnes