govorox/SSLClient #81: Failure to send POST requests over with TinyGSM

reusables-official, 2024-05-22:
Hello, I am observing repeated failures trying to send a post request via SSLClient using my SIMCOM7600G modem with TinyGSM.

The most common and consistent error message:
```
[318168][E][ssl__client.cpp:401] init_tcp_connection(): Connection to server failed, is the signal good, server available at this address and timeout sufficient? reusables.hyperether.com:443
[319175][E][ssl__client.cpp:46] _handle_error(): [start_ssl_client():371]: (-2) BIGNUM - An error occurred while reading from or writing to a file
[319177][E][SSLClient.cpp:242] connect(): start_ssl_client failed: 0
```

I have tried to set headers to adjust the time out like so:

```
client.connectionKeepAlive();
client.setHttpWaitForDataDelay(5000);
client.sendHeader("Keep-Alive", "timeout=1000, max=5000");
int err = client.post(API_ENDPOINTS.container, "application/json", body);
```

where body is a relatively small json payload:
```
{
  "mode": "return-containers",
  "data": {
    "containerIds": [
      "64367339246bf8633c782348"
    ],
    "locationId": "6621a0099cf68973c58f68bf"
  }
}
```

Here is my setup code:

```
#define TINY_GSM_MODEM_SIM7600
#define TINY_GSM_RX_BUFFER 2048

#include <TinyGsmClient.h>
#include <ThingerTinyGSM.h>

HardwareSerial gsmSerial(1);
TinyGsm modem(gsmSerial);
TinyGsmClient gsmClient(modem, 0);

SSLClient secure_layer_lte(&gsmClient);
HttpClient lte_ssl_client = HttpClient(secure_layer_lte, API_HOST, PORT);

void lteInit() {
  gsmSerial.begin(BAUD_RATE, SERIAL_8N1, LTE_RX, LTE_TX, false);
  secure_layer_lte.setCACert(api_root_ca);
  modem.init();
  modem.setNetworkMode(38);
  modem.sendAT("AT+CMNB=1");
  modem.gprsConnect(APN);
};
```

Then I use a RTOS task to monitor the connection and reconnect as needed:

```
void lteConnect(void * parameter) {
  lteInit();
  for (;;) {
    if (modem.isNetworkConnected()) {
      if (modem.isGprsConnected()) {
        vTaskDelay(20000 / portTICK_PERIOD_MS);
      } else {
        modem.gprsConnect(APN);
      };
    } else {
      bool networkConnected = modem.waitForNetwork();
    };
    vTaskDelay(5000 / portTICK_PERIOD_MS);
  };
  vTaskDelete(NULL);
};
```

What confuses me the most is that sometimes the request will go through and I'll get a 200 back as if nothing is wrong, but the vast majority of the time, I get the error pasted above.

I have run a test on signal quality to assess, and I will get a mix of 99 back as a value and anything from 28-31 at other points. Running on an interval of one second might produce output like: `[99, 31, 31, 31, 99, 27, 99, 99, 99, 28]` which also confuses me. My understanding is that a value of `99` for signal quality from TinyGSM means it could not retrieve the value, 31 is the best quality possible and that a value of 0 would mean there was no reception at all.

Any ideas what I'm doing wrong or inconsistently? I know the error is being thrown by this library, but very possible some other culprit is to blame here. I have the same functionality working perfectly over WiFi, where I provide the WiFi client to the "client.post" function I pasted above, also with SSL also via this library.

Very puzzled. Any thoughts greatly appreciated. I commented on a recent PR for this library that I thought might address the issue but it has not been resolved in my case, I am on version 1.2.0.

Platform.ini

```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
lib_deps = 
	bblanchon/ArduinoJson@7.0.3
	vshymanskyy/TinyGSM@0.11.7
	makuna/NeoPixelBus@2.7.8
	arduino-libraries/ArduinoHttpClient@0.6.0
	digitaldragon/SSLClient@1.2.0
	thinger-io/thinger.io@2.27.0
	bblanchon/StreamUtils@1.8.0
```

reusables-official, 2024-05-22:
Spent the day trying to tweak things, running examples from this repository. Still seeing totally variable results, some calls succeed, most fail. I've tried interacting with different servers like the `vsh.pp.ua` from tiny GSM repo to the same effect. Using a client without SSL seems to have the same effect so, really, I don't think the issue I'm seeing is rooted in this repository.

kgmuzungu, 2024-09-16:
I see something similar on a SIMCom A7670E modem with a ESP32. If I restart the modem, e.g. every 30s, to send a bunch of IoT data, then it works fine. But if I try to keep the connection alive, some HTTPS POST work, but mostly not.

The worst of the sslClient error is that it blocks until, my guess, some underlying TCP/IP stack function times out. That takes some 30sec. It would be great being able to set a shorter timeout!

Or better a fix. 

@reusables-official did you get any further?

reusables-official, 2024-09-17:
@kgmuzungu Interesting theory. We moved on from using this SSL Client and instead were using a fork of the TinyGSM library that had "native" SSL support. I think there is a link to it in the issues of TinyGSM repository. 

We also eventually stopped doing that and offloaded all of our networking on to our IoT cloud provider (thinger.io) and generally find this much more reliable.

kgmuzungu, 2024-09-18:
Thank you for the hints. Native SSL is not working with all firmware versions of the SIMComA7670E.... and I havnt found how to upgrade/change the firmware. 

I am still looking for possiblities to get SSLClient working with A7670E 🙏

RobertByrnes, 2026-11-02:
The original log identifies a failure before TLS starts: init_tcp_connection() reports that the underlying TinyGsmClient::connect(host, 443) failed. SSLClient returns its own -2 for that condition; passing it through the mbedTLS error formatter produces the misleading BIGNUM description. That wording does not establish a cryptographic or file-access failure.

The original reporter subsequently found similar failures with a non-SSL client, which is important evidence that their particular failure was below the TLS layer. The later A7670E and SIM7080 reports may involve different causes and should not yet be treated as confirmed duplicates.

For anyone still reproducing this, please retest with SSLClient 1.4.0 and provide exact ESP32 core, TinyGSM, modem firmware, and HTTP-client versions. Please isolate the layers:

1. Run a minimal, single-task test of plain TinyGsmClient TCP connection to the same host and port. Stop that test connection afterwards. Record the result and relevant AT exchange. If this fails, investigate modem/network/DNS/socket handling before changing certificates or TLS settings.
2. If TCP works, test SSLClient connection separately with the correct CA and a correctly initialized ESP32 system clock, capturing verbose SSL logs.
3. Only then add the HTTP POST and connection reuse. Capture the post() return value separately from responseStatusCode(), and consume the response or stop the HTTP client before starting another request.

A few specific points in the posted code:

- TinyGSM's sendAT() already prepends AT. sendAT("AT+CMNB=1") therefore sends a duplicated AT prefix. Only issue commands supported by your exact modem/firmware, pass the command suffix to sendAT(), and consume/check its response before another modem operation. This is a concrete sketch issue, not proof it caused every reported failure.
- Check the results of modem.init(), network configuration, and gprsConnect(), rather than continuing after failed initialization.
- The monitoring task also issues modem commands and may reconnect GPRS. If HTTP uses the same modem from another task, those operations can interfere. First reproduce with one owner/task for all modem/UART operations; do not reconnect the data context during an active request.
- connectionKeepAlive() and an HTTP Keep-Alive header do not configure the modem's TCP connect timeout. setHttpWaitForDataDelay() controls the HTTP polling delay, not that timeout. SSLClient's current init_tcp_connection() calls the underlying two-argument connect(); its setTimeout() is not a guaranteed end-to-end deadline for the modem connection attempt.
- If the later -3 is returned by ArduinoHttpClient, it means HTTP_ERROR_TIMED_OUT, not automatically a TinyGSM error. Please identify the method that returned each code.

For the SIM7080 +CAURC: "buffer full" report, please include a separate minimal reproducer and AT trace showing the receive/drain sequence, firmware, TinyGSM version, and whether plain TCP also reproduces it. That deserves its own transport-specific investigation rather than assuming the SIM7600 diagnosis applies.

References: [TinyGSM sendAT implementation](https://github.com/vshymanskyy/TinyGSM/blob/v0.11.7/src/TinyGsmModem.tpp), [ArduinoHttpClient error definitions](https://github.com/arduino-libraries/ArduinoHttpClient/blob/master/src/HttpClient.h), and [SSLClient TCP initialization](https://github.com/govorox/SSLClient/blob/1.4.0/src/ssl__client.cpp).

Please redact credentials, tokens, SIM/device identifiers, and private endpoint details from logs. Keeping this open with help wanted pending a current minimal reproducer; we are not claiming the newer release resolves these modem-specific failures.