govorox/SSLClient #83: Will ESP32 core 3.0.0 MBEDTLS issue be an issue going forward?

richardbarkeriom, 2024-06-01:
I have successfully been using SSLClient in my code for some months. However, Expressif has recently released core v3.0.0 which appears to have withdrawn MBEDTLS. I have for now reverted to core 2.0.17 which works fine but are there any plans to work around the MBEDTLS issue?

duyle1402, 2024-06-04:
I got the same error as you when updating to ***arduino esp32 core 3.0.0***. Is there any way to solve this problem?

richardbarkeriom, 2024-06-04:
I reverted my core to 2.0.17 and my code now compiles. I'm not sure what
our options are going forward.

anthonyhoegberg, 2024-07-03:
I looked into this myself and also looked at the newer secure network implementation [NetworkClientSecure](https://github.com/espressif/arduino-esp32/tree/master/libraries/NetworkClientSecure) to see what needed to be changed to possibly add support for the new core. However im not sure what needs to be done to implement support for v3 any ideas?

RobertByrnes, 2024-07-03:
Hi @richardbarkeriom @duyle1402  @anthonyhoegberg  @Bascy I have just had a dig through this and here is what I have found out so far:

The latest version of the PlatformIO Core is `6.1.15`, which includes various improvements and bug fixes​ ([GitHub](https://github.com/platformio/platformio-core/releases))​​ ([PlatformIO Labs](https://piolabs.com/blog/news/platformio-oss-may-2024-updates.html))​.

For the ESP32 platform (espressif32), the latest version is `6.7.0`. This version includes updates for new boards and support for Arduino framework version `2.0.16`, along with several enhancements in the toolchain and debugging options​ ([PlatformIO Labs](https://piolabs.com/blog/news/platformio-oss-july-2022-updates.html))​​ ([GitHub](https://github.com/platformio/platform-espressif32/releases))​.

The latest version of the framework-arduinoespressif32 is `3.0.0`. This release is based on ESP-IDF `v5.1.4` and introduces support for new SoCs like ESP32-H2 and ESP32-C6, along with other significant updates​ ([GitHub](https://github.com/espressif/arduino-esp32/releases))​​ ([GitHub](https://github.com/espressif/arduino-esp32))​.

Having PlatformIO installed in a windows machine I have the mbedtls library version at this location:

`C:\Users\<MY_USERNAME>\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\include\mbedtls\mbedtls\include\mbedtls\version.h` - This specifies the version of mbedtls as 2.28.7

anthonyhoegberg, 2024-07-03:
Maybe rewriting SSLClient from scratch using the new NetworkClientSecure code as a base could be a start? Adapting it for 3.x MbetTLS api?

RobertByrnes, 2024-07-05:
Here's a summary regarding the Mbed TLS version in different frameworks and the issues encountered with PlatformIO:

## Mbed TLS Version:
- The Arduino IDE pulls in Mbed TLS version 3.6.0 via the IDF release 5.1-bd2b9390ef.
- PlatformIO using the ESP-IDF framework version 3.40407.240606, which includes Mbed TLS 3.6.0.
- The framework-arduinoespressif32 version 3.20017.0 in PlatformIO corresponds to tag 2.0.17 in the GitHub repository [arduino-esp32](https://github.com/espressif/arduino-esp32/tree/master) - which only includes mbedtls 2.28.7.

## Version Discrepancies:
PlatformIO uses its own versioning scheme which may not always align with the GitHub repository tags.
The Arduino IDE package manager pulls later versions compared to what is available in PlatformIO.

## Error in PlatformIO:
When trying to update the PlatformIO configuration to use the latest code from GitHub, a KeyError: 'framework-arduinoespressif32-libs' error occurred. This suggests missing components expected by PlatformIO.
Attempted solutions included reverting to the stable release version, ensuring proper submodules are initialized, and specifying a particular stable tag instead of the master branch.

## Suggestions for Resolution:

Use the stable release version in PlatformIO configuration.
Ensure all Git submodules are properly initialized.
Specify a stable tag in the platformio.ini file.
Check for and include any missing dependencies.
- Having tried this it didn't work!

richardbarkeriom, 2024-07-12:
> @RobertByrnes <https://github.com/RobertByrnes> since PlatofrmIO doesn't
> officially support arduino-esp32 v3, you have to manually gather both the
> base, and the libs from Espressif's github repository. It should be easy to
> do, you just need to make sure both the framework and the libs are imported
> into the project. You need to add these into the platformio.ini file:
>
> platform = espressif32
> framework = arduino
> platform_packages=
>   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.2
>   framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.2/esp32-arduino-libs-3.0.2.zip
> build_flags =
> 	-std=gnu++2a
> 	-std=c++11
> 	-std=gnu++11
> build_unflags = -std=gnu++2b
>
> This ensures all dependecies needed for arduino-esp32 are installed.
> Also, I have noticed the differences between v2 and v3 are not that
> significant in terms of MbedTLS. This should be quite easy to patch,
> however it could lead to the library not being compatible with the former
> once adapted to the latter.

That's good to know. Would it be too much to ask to make a "legacy (core
v2)" version available too? This is the "cake and eat it" option while
making sure everything runs OK under the v3 core. I'm personally planning
to run up two installs for the time being as I have a large project I'm
working on that I don't want breaking!

RobertByrnes, 2024-07-12:
@richardbarkeriom you can rest easy! or rather enjoy working on your project. We took care of it!

closed 2024-07-12 by RobertByrnes