govorox/SSLClient #89: Error in compilation "mbedtls/platform.h: No such file or directory"

Seikonolff, 2024-07-10:
Hi guys,

I am trying to use the lib on an arduino nano 33 IoT on platformIO.

I have the following depencies :
`
[env:nano_33_iot]
platform = atmelsam
board = nano_33_iot
framework = arduino
lib_deps = 
	digitaldragon/SSLClient@^1.2.0
	droscy/esp_mbedtls_esp8266@^2.22300.2
`

But when I try to compile the project I get stuck with the following error :

.pio\libdeps\nano_33_iot\SSLClient\src\certBundle.c:16:10: fatal error: esp_system.h: No such file or directory

Do you have any idea of how I could fix this ?

Thanks a lot for your help,

Mathis

RobertByrnes, 2024-07-10:
Hi @Seikonolff This is interesting as I have no experience of the `atmelsam` platform but it is the `espressif32` platform which pulls in the mbedtls library version which this library is built on top of. Instead of using the `droscy/esp_mbedtls_esp8266@^2.22300.2` library, perhaps try this one instead:

```ini
https://github.com/ARMmbed/mbedtls.git#v2.28.7
``` 

I don't know if this will work for your platform but it is the correct version for the current state of the SSLClient.

Seikonolff, 2024-07-10:
Hi Robert,

It's maybe cause of the 'atmelsan' platform. Even with the lib from the repo I got these errors :

.pio\libdeps\nano_33_iot\SSLClient\src\ssl__client.h:12:10: fatal error: mbedtls/platform.h: No such file or directory
 #include <mbedtls/platform.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
.pio\libdeps\nano_33_iot\SSLClient\src\certBundle.c:16:10: fatal error: esp_system.h: No such file or directory

here is the repo of my code : https://github.com/Seikonolff/tinysGsmTest

PR #100 (2024-07-24): Hotfix - 95 release 130 fails to compile on arduino esp32 v3
closes #89 
closes #95
closes #96 
closes #99
closed 2024-07-24 by RobertByrnes