govorox/SSLClient #93: mbedtls/net.h issue

DhruvChugh, 2024-07-14:
Hi
Decription: MbedTLS.h and function emulator.h file not found
Expected Behaviour: Finds the files and compiles
Actual Behaviour:
\Arduino\libraries\GovoroxSSLClient\src/ssl__client.h:15:10: fatal error: mbedtls/net.h: No such file or directory
15 | #include <mbedtls/net.h>
| ^~~~~~~~~~~~~~~
compilation terminated.

Note: I have installed the latest version if this library
IDE: Arduino IDE
Board : ESP32 dev module
GSM : sim800

RobertByrnes, 2024-07-14:
Hi @DhruvChugh This is a bug I think. Not in the library but in the examples. Apologies for this problem. This will work in VSCode IDE with PlatformIO extenstion - there is a learning curve to follow if you try this.

I will be looking at the Arduino IDE examples over the next week. I plan on having this fixed in the next release which will be v1.3.0 at the end of this working week.

GotRobbd, 2024-07-16:
@DhruvChugh what version of arduino-esp32 are you using? If you're using v3, you need to wait for the next release to be out, as it's not currently compatible with it. In the meantime, you should use arduino-esp32 v2.

DhruvChugh, 2024-07-16:
@GotRobbd Hi ,i didn’t understand what you meant by arduino-esp32 v3/v2 .
 I am using the latest version of Arduino IDE. The microcontroller i am using has an ESP32 Wroom chip.

RobertByrnes, 2024-07-16:
@DhruvChugh He means the latest version of the ESP32 libraries for Arduino platform is at version 3.0.2 (you can view this in the Boards Manager of Arduino IDE) The older major version is 2.0.17.

@GotRobbd The new version of this library does not solve that. There are compilation error which I am looking into at the moment.

CanselYalnkaya, 2024-07-17:
Hi everyone, 

I have been dealing with this error for days. I am also getting the following error:

In file included from /Users/firefly/Documents/Arduino/libraries/GovoroxSSLClient/src/SSLClient.h:23,
                 from /Users/firefly/Downloads/main 2/main/main.ino:9:
/Users/firefly/Documents/Arduino/libraries/GovoroxSSLClient/src/ssl__client.h:15:10: fatal error: mbedtls/net.h: No such file or directory
   15 | #include <mbedtls/net.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

IDE: Arduino IDE 2.2.1
Board : ESP32 dev module 3.0.1
GSM : SIM7600G-H

The code was compiled on a computer with Windows 10 using Arduino IDE. However, despite trying on 5 other computers, it did not compile on them. What do you recommend for this issue?

RobertByrnes, 2024-07-17:
Hi @CanselYalnkaya For now open Arduino IDE - go to Boards Manager - type ESP32 and downgrade ESP32 by Espressif to `v2.0.17`.

This library will have `v1.3.0` published within the week which give compatibility for ESP32 by Espressif >= v3.0.0.

There is catch! This will all work fine in VSCode with PlatformIO but not in Arduino IDE. There is breaking change for Arduino IDE from ESP32 by Epsressif from `v3.0.0`. We may not have a fix ready for `v1.3.0` of this library but will endever to have on out as soon as possible after.

CanselYalnkaya, 2024-07-19:
Thank you so much @RobertByrnes. Problem is solved with ESP version 2.0.17 and Platform IO.

RobertByrnes, 2024-07-24:
@DhruvChugh @CanselYalnkaya I am about to make release 1.3.1 - please do try this. It 'should' now work accross PlatformIO and Arduino IDE for ESP 2.0.17 and 3.0.2

DhruvChugh, 2024-07-24:
Thank you @RobertByrnes , will test it out and inform the result here.

RobertByrnes, 2026-11-02:
The mbedTLS header compatibility changes are now released. Please use GovoroxSSLClient/SSLClient 1.4.0: https://github.com/govorox/SSLClient/releases/tag/1.4.0

The current code selects mbedtls/net_sockets.h for mbedTLS 3 in the normal configuration, and the older mbedtls/net.h path for mbedTLS 2. Release 1.4.0 passed Arduino example builds against ESP32 Arduino cores 2.0.17 and 3.0.2, as well as the PlatformIO example matrix.

To clarify the version terminology in the earlier discussion: the Arduino IDE version and the ESP32 board-package version are separate. In Arduino IDE, inspect the installed esp32 package by Espressif Systems in Boards Manager. Please also ensure Library Manager has GovoroxSSLClient 1.4.0 installed and check the compiler output for duplicate/older library copies.

MbedTLS.h and the emulator headers mentioned in the original description are repository test-only dependencies. Normal firmware builds use the mbedTLS headers provided by the ESP32 core and should not require installing those mock headers. Do not enable SSL_CLIENT_TEST_ENVIRONMENT for a firmware build. Also remove an old manually added MBEDTLS_BACKPORT flag unless your specific toolchain requires that compatibility override; it changes the header/API selection.

Closing as completed based on the released changes and build coverage. If the missing-header error recurs, please follow up with the exact ESP32 Boards Manager package version, SSLClient version, full compiler output (including the selected library path), and any custom build flags. The tested versions above are not a claim that every possible core/toolchain combination has been validated.

closed 2026-11-02 by RobertByrnes