govorox/SSLClient #94: add compatibility to ESP-IDF

anthonyhoegberg, 2024-07-22:
Could we add a cmakelist file for the project so we can include it as a component in esp-idf based projects that does not use platformio or arduino ide

RobertByrnes, 2024-07-26:
Hi @anthonyhoegberg Yes this is a great idea. Have you done this yourself with this code base? If so, what could you share / pr to make this happen? Any help with this appreciated.

anthonyhoegberg, 2025-09-19:
I tried my best on this at the time for a few weeks and failed horribly sorry

RobertByrnes, 2026-11-02:
There are two possible scopes here, and adding CMakeLists.txt alone would not make the current library Arduino-free:

1. ESP-IDF with Arduino-ESP32 included as a component. This would allow an idf.py/CMake-based project without Arduino IDE or PlatformIO, while retaining SSLClient's Arduino Client transport API. Espressif supports that build model: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html
2. Pure ESP-IDF without an Arduino dependency. That needs a broader port because the current implementation depends on Client, IPAddress, Arduino timing functions, and logging facilities.

We have opened a branch to explore the first route: https://github.com/govorox/SSLClient/tree/codex/94-esp-idf-arduino-component

This is an investigation branch, not yet an implemented or validated compatibility claim. The intended work is component registration, explicit Arduino/IDF dependencies, a minimal example, and a reproducible build check against a compatible pinned Arduino/ESP-IDF pair.

Would an Arduino-as-component dependency meet your needs? Please also share your target ESP-IDF version, ESP32 variant, and underlying transport (for example, Wi-Fi, Ethernet, or TinyGSM), and any build errors from your previous attempt. That will help establish a useful compatibility target.

Keeping this open as an enhancement. Thanks for reporting back about your earlier attempt; the existing Arduino dependencies make this more than just a missing build file.