govorox/SSLClient #95: Release 1.3.0 fails to compile on arduino-esp32 v3

GotRobbd, 2024-07-22:
The recent update is great, it manages to abide by the new MbedTLS version pushed by Espressiff for their new Arduino release. However, this no longer compiles on PlatformIO using either versions 3.0.2 or 3.0.3. The compilation errors look like this:
```
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:968:30: error: 'string' does not name a type; did you mean 'String'?
  968 | static bool match_name(const string& name, const string& domainName) {
      |                              ^~~~~~
      |                              String
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:968:50: error: 'string' does not name a type; did you mean 'String'?
  968 | static bool match_name(const string& name, const string& domainName) {
      |                                                  ^~~~~~
      |                                                  String
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp: In function 'bool match_name(const int&, const int&)':
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:969:29: error: request for member 'find' in 'name', which is of non-class type 'const int'
  969 |   size_t wildcardPos = name.find("*");
      |                             ^~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:973:22: error: 'string' has not been declared
  973 |   if (wildcardPos == string::npos) {
      |                      ^~~~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:978:29: error: request for member 'find' in 'name', which is of non-class type 'const int'
  978 |   size_t firstDotPos = name.find(".");
      |                             ^~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:991:45: error: request for member 'find' in 'domainName', which is of non-class type 'const int'    
  991 |   size_t domainNameFirstDotPos = domainName.find('.');
      |                                             ^~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:992:32: error: 'string' has not been declared
  992 |   if (domainNameFirstDotPos == string::npos) {
      |                                ^~~~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:995:21: error: request for member 'substr' in 'domainName', which is of non-class type 'const int'  
  995 |   return domainName.substr(domainNameFirstDotPos) == name.substr(firstDotPos);
      |                     ^~~~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:995:59: error: request for member 'substr' in 'name', which is of non-class type 'const int'        
  995 |   return domainName.substr(domainNameFirstDotPos) == name.substr(firstDotPos);
      |                                                           ^~~~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp: In function 'bool verify_ssl_dn(sslclient__context*, const char*)':
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:1090:3: error: 'string' was not declared in this scope
 1090 |   string domain_name_str(domain_name);
      |   ^~~~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:14:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
   13 | #include "certBundle.h"
  +++ |+#include <string>
   14 |
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:1091:13: error: 'domain_name_str' was not declared in this scope; did you mean 'domain_name'?       
 1091 |   transform(domain_name_str.begin(), domain_name_str.end(), domain_name_str.begin(), ::tolower);
      |             ^~~~~~~~~~~~~~~
      |             domain_name
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:1099:11: error: expected ';' before 'san_str'
 1099 |     string san_str((const char*)san->buf.p, san->buf.len);
      |           ^~~~~~~~
      |           ;
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:1100:15: error: 'san_str' was not declared in this scope
 1100 |     transform(san_str.begin(), san_str.end(), san_str.begin(), ::tolower);
      |               ^~~~~~~
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:1117:13: error: expected ';' before 'common_name_str'
 1117 |       string common_name_str((const char*)common_name->val.p, common_name->val.len);
      |             ^~~~~~~~~~~~~~~~
      |             ;
.pio/libdeps/esp32wroom/SSLClient/src/ssl__client.cpp:1119:22: error: 'common_name_str' was not declared in this scope; did you mean 'common_name'?
 1119 |       if (match_name(common_name_str, domain_name_str)) {
      |                      ^~~~~~~~~~~~~~~
      |                      common_name
*** [.pio\build\esp32wroom\lib30d\SSLClient\ssl__client.cpp.o] Error 1
```
It does seem like ssl__client.cpp is trying to use the C++ string, which is not included in the file as a library. Adding #include <string> into it fixes the issue. Is there anything else that can be done?

GotRobbd, 2024-07-22:
It seems like the library was removed from the file in commit cbc8d3cb2ac2d792fc102f41b45807d9caed6d9c whilst working on #91 pull request.

RobertByrnes, 2024-07-23:
Hi @GotRobbd be great if you could try this branch: '95-release-130-fails-to-compile-on-arduino-esp32-v3'. I think the PlatformIO side of things is fixed. The workflow now incorporates a run for v3.0.2 as well as the legacy.

PR #100 (2024-07-24): Hotfix - 95 release 130 fails to compile on arduino esp32 v3
closes #89 
closes #95
closes #96 
closes #99
closed 2024-07-24 by RobertByrnes