govorox/SSLClient #96: Discussion: regarding Arduino IDE example code using arduino-esp32 v3

GotRobbd, 2024-07-22:
I have notice that people may stumble upon using the library properly when using the latest version of arduino-esp32 (currently at 3.0.3). This has been caused by an external library named [ArduinoHttpClient](https://github.com/arduino-libraries/ArduinoHttpClient.git), which fails to build. 

This is due to the GCC compiler's byte class and the Arduino's byte typedef classhing with each other, causing ambiguity. This is not an issue when using arduino-esp32 v2.0.17 and lower, however. This problem is arising now because arduino-esp32 v3 is using GCC 12.2.0, whilst arduino-esp32 v2 is using is using GCC 8.4.0. This is important because the former is using the C++17 standard or higher, which introduce the byte class in the language. The latter is still using C++11, which does not have such class.

The problem not only affects ArduinoHttpClient, but any other Arduino library using the byte typedef, including [TinyGsm](https://github.com/vshymanskyy/TinyGSM.git).

The good news is that there is an easy workaround. When opening one of the examples in ArduinoIDE from this library, look for ```#include <ArduinoHttpClient.h> ``` and put it as the first include. This should fix the issue and the code should compile just fine. If it still does not work, try and put it in between the other includes until the code compiles.

Original https_wifi example code in ArduinoIDE, fails to compile due to "byte" definition ambiguity:
[image]

Modified https_wifi example code in ArduinoIDE, compiles fine:
[image]

RobertByrnes, 2024-07-24:
Hi @GotRobbd About to make 1.3.1 release - please try it. Should fix both PlatformIO and Arduino IDE for both 2.0.17 and 3.0.2

closed 2024-07-24 by RobertByrnes

duyle1402, 2024-07-24:
Hi @RobertByrnes, @GotRobbd 

The error `reference to 'byte' is still ambiguous` in the Arduino example regarding the `TinyGsm` library. Does anyone have any ideas on how to debug this issue? Should we raise an issue in the `arduino-esp32` and `TinyGsm` repositories to include information about the ambiguity between the new GCC byte class and Arduino's byte `typedef` in the ESP32 migration guide?

RobertByrnes, 2024-07-24:
Hi @duyle1402 Which version of this lib? `1.3.0`?

duyle1402, 2024-07-24:
Nevermind, sorry for the confusion. I see that you used `using namespace std;` in release `1.3.1`. I just tested it again and it works well. Sorry again @RobertByrnes

RobertByrnes, 2024-07-24:
Hey @duyle1402 No trouble - just glad it is sorted!

PR #100 (2024-07-24): Hotfix - 95 release 130 fails to compile on arduino esp32 v3
closes #89 
closes #95
closes #96 
closes #99