govorox/SSLClient #98: Issue with a connection MQTT over TLS with Ethernet W5500

MaxiLargo, 2024-07-22:
### Description

### Steps to Reproduce

```
boolean reconnect() {
  chipId = ESP.getEfuseMac();
  char chipIdString[18];  
  snprintf(chipIdString, sizeof(chipIdString), "%04X%08X", (uint16_t)(chipId >> 32), (uint32_t)chipId);

  if (client.connect(chipIdString, mqtt_username, mqtt_password)) {
    client.publish("test", message.c_str());
    conectado.store(true);
  }
  return client.connected();
}

void mqttTask(void* param) {
  while (1) {

unsigned long cu = millis();

if (cu - lastReconnectAttempt >= 60000) {
 
    String message = "....";

    if (!client.connected()) {
        reconnect();
      }else {
        client.loop();
    }

  }

  unsigned int highWaterMark = uxTaskGetStackHighWaterMark(NULL);
  Serial.print(highWaterMark);

  lastReconnectAttempt = cu;
}
  vTaskDelay(pdMS_TO_TICKS(60000));
}
}

static const char* root_ca PROGMEM = R"EOF(
...
)EOF";

EthernetClient ethclient;
SSLClient seceth(&ethclient);
PubSubClient client(seceth);

void setup() {
  WiFi.mode(WIFI_STA); 
  Serial.begin(115200);
  disableCore0WDT();
  disableCore1WDT();
  M5.begin();        
  M5.Power.begin();
    
  uint64_t chipId1 = 0;
  chipId1 = ESP.getEfuseMac();

  char chipIdString[18];
  snprintf(chipIdString, sizeof(chipIdString), "%04X%08X", (uint16_t)(chipId1 >> 32), (uint32_t)chipId1);
  Serial.print(chipIdString);
  String ssid = "indulink" + String(chipIdString);
  
  if (!SD.begin()) {
        M5.Lcd.println("Card failed, or not present");
        while (1);
    }
    M5.Lcd.println("TF card initialized.");

    while (!module.begin(&Wire, 21, 22, MODULE_4IN8OUT_ADDR)) {
        Serial.println("4IN8OUT INIT ERROR");
        M5.Lcd.println("4IN8OUT INIT ERROR");
    }
    
    Serial.println("4IN8OUT INIT SUCCESS");

    M5.Lcd.println("Contador de pulsos iniciado.");

  SPI.begin(SCK, MISO, MOSI, -1);
  Ethernet.init(CS);
    while (!Serial) { }

  Ethernet.begin(mac);
  Serial.print("Ethernet iniciado. Dirección IP: ");
  Serial.println(Ethernet.localIP());
  timeClient.begin();
  timeClient.update();
  seceth.setCACert(root_ca);

  client.setServer(mqtt_server, mqtt_port);
  client.setCallback(callback);
  Reinicio.attach(2,reinicio);

  xTaskCreatePinnedToCore(
    mqttTask,    
    "mqttTask",  
    8192,         
    NULL,
    1, 
    NULL,
    0   
  );
  xTaskCreatePinnedToCore(
    countTask,   
    "countTask", 
    8192,       
    NULL,
    1,  
    NULL,
    1   
  );
}
```
### Expected Behavior

Publish a variable in a broker mqtt over tls with ethernet
### Actual Behavior

when the mqtt task run, this error appears
>[ 62514][E][ssl__client.cpp:42] _handle_error(): [start_ssl_client():367]: (-1) ERROR - Generic error
>[ 62515][E][SSLClient.cpp:242] connect(): start_ssl_client failed: 0

### Environment

- **Operating System:** [Windows 10 Home]
- **Version:** [22H2]

### IDE

- [x] Arduino IDE
- [ ] PlatformIO IDE extension

### Additional Context

It is a very big program, try to show the functions that are used by the mqtt task

RobertByrnes, 2024-07-26:
Hi @MaxiLargo Have you got anywhere with this?

1. Be good to see the output when log level set to debug.
2. Have you set the flag for the

MaxiLargo, 2024-07-29:
Hi @RobertByrnes!
I try to set the debug level (`seceth.setDebugStream(&Serial)`) but it doesn't work. And yes, I have the flag `#define W5500_WORKAROUND` and also I tried with `#define _W5500_H_` but it is the same result, i have the last version of the lib.

If you can help me I would appreciate it

RobertByrnes, 2024-07-29:
ok @MaxiLargo can you share a longer excerpt from log file that shows the entire connection process leading up o these 2 errors:

```bash
[ 62514][E][ssl__client.cpp:42] _handle_error(): [start_ssl_client():367]: (-1) ERROR - Generic error
[ 62515][E][SSLClient.cpp:242] connect(): start_ssl_client failed: 0
```

MaxiLargo, 2024-07-30:
Here it's the log:
> entry 0x400805f0
> [    M5Stack initializing...
> OK
> [   410][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
> Ethernet init.IP: 192.168.0.106
> C88CA51FB6084IN8OUT INIT SUCCESS
> [  2112][E][ssl__client.cpp:45] _handle_error(): [start_ssl_client():370]: (-1) ERROR - Generic error
> [  2112][E][SSLClient.cpp:242] connect(): start_ssl_client failed: 0

songhellobulter, 2025-01-06:
Has this problem solved? I have the same error with w5500 too.
[image]

RobertByrnes, 2026-11-01:
This report overlaps with the W5500 connection problem in #115 and the Ethernet idle-read behaviour addressed in #120. It is separate from #110, which concerns large Telegram photo uploads over SIM800L and uses SSLClientESP32 in the posted sketch.

Please retest with GovoroxSSLClient/SSLClient 1.3.3: https://github.com/govorox/SSLClient/tree/1.3.3

The receive callback now translates negative reads into MBEDTLS_ERR_SSL_WANT_READ while the underlying client remains connected. This handles Ethernet-style clients returning -1 when no data is waiting, without requiring the old W5500_WORKAROUND define. See PR #121: https://github.com/govorox/SSLClient/pull/121

That is relevant to the generic -1 connection error reported here, but we have not confirmed the cause on your hardware. Please let us know whether MQTT connects and remains connected while idle after updating.

If it still fails, please provide the Ethernet library/version, ESP32 Arduino core version, a minimal reproducer including timeout calls, and verbose logs covering the entire connection attempt. In Arduino IDE select Tools > Core Debug Level > Verbose; setDebugStream() is not an API provided by this SSLClient library. Include SSLClient::lastError() after a failed connection, and redact credentials and private endpoint details.

Also confirm the ESP32 system clock is set correctly before TLS certificate validation. If timeClient is NTPClient, calling timeClient.update() updates that object's time; it does not by itself set the ESP32 system clock used by certificate verification.

Keeping this issue open with help wanted pending test results. The overlap is not yet sufficient to close it as a confirmed duplicate or resolved issue.