govorox/SSLClient #99: getting compilation error with the example https_post_sim7600

Manojkumar0308, 2024-07-23:
|             domain_name
c:\Users\Acer\Documents\Arduino\libraries\GovoroxSSLClient\src\ssl__client.cpp:1099:11: error: expected ';' before 'san_str'
 1099 |     string san_str((const char*)san->buf.p, san->buf.len);
      |           ^~~~~~~~
      |           ;
c:\Users\Acer\Documents\Arduino\libraries\GovoroxSSLClient\src\ssl__client.cpp:1100:15: error: 'san_str' was not declared in this scope
 1100 |     transform(san_str.begin(), san_str.end(), san_str.begin(), ::tolower);
      |               ^~~~~~~
c:\Users\Acer\Documents\Arduino\libraries\GovoroxSSLClient\src\ssl__client.cpp:1117:13: error: expected ';' before 'common_name_str'
 1117 |       string common_name_str((const char*)common_name->val.p, common_name->val.len);
      |             ^~~~~~~~~~~~~~~~
      |             ;
c:\Users\Acer\Documents\Arduino\libraries\GovoroxSSLClient\src\ssl__client.cpp:1119:22: error: 'common_name_str' was not declared in this scope; did you mean 'common_name'?
 1119 |       if (match_name(common_name_str, domain_name_str)) {
      |                      ^~~~~~~~~~~~~~~
      |                      common_name

RobertByrnes, 2024-07-24:
Hi @Manojkumar0308 I about to make the release for version 1.3.1 which 'should' fix the issue you were having. Please try this new version.

Manojkumar0308, 2024-07-24:
after how many hours or minutes i will try  please tested it well and give
a working code as an example that i will go through to it.

RobertByrnes, 2024-07-24:
For PlatformIO I think it will be there within the our usually. For Arduino IDE anytime up to about 24 hours...

Manojkumar0308, 2024-07-24:
Ok, thanks for letting me know please consider my point of testing and
giving sample code to install SSL Certificate on module

PR #100 (2024-07-24): Hotfix - 95 release 130 fails to compile on arduino esp32 v3
closes #89 
closes #95
closes #96 
closes #99