govorox/SSLClient #103: SIM7600G Unhandled: +CIPCLOSE: 0,0 (Error code -2)

AltayAtaman, 2024-08-13:
**Description:** I'm trying to connect my SIM7600 to Azure IoT over cellular and secure connection is needed for establishing connection with Azure IoT. Because of this requirement I'm trying to use this library to add a secure layer. But I was not able to even make example code work with public broker provided in the example. I only changed the GSM initialization part to make it compatible with my SIM7600G but the code is not working. It's giving "Attempting MQTT connection...[147598] ### Unhandled: +CIPCLOSE: 0,0 failed, rc=-2...try again in 15 seconds" as error.

I have seen the other topic opened up which might have exactly the same issue but the GSM used in that project is different the mine and the solution provided (which is changing the certificate as far as I understood) didn't work for me. 

**Steps to Reproduce:** Connect ESP32 to SIM7600 and upload the following code: 

```
#include "SSLClient.h"
#include <PubSubClient.h>

#define TINY_GSM_DEBUG Serial

#include "ca_cert.h"

// SIM7600 defines added by me:
#define ESP32_TYPE 3
#define GSM_ENABLED 1
#define uS_TO_S_FACTOR      1000000ULL  /* Conversion factor for micro seconds to seconds *
#define TIME_TO_SLEEP       30          /* Time ESP32 will go to sleep (in seconds) */
#define UART_BAUD           115200
#define MODEM_TX            27
#define MODEM_RX            26
#define MODEM_PWRKEY        4
#define MODEM_DTR           32
#define MODEM_RI            33
#define MODEM_FLIGHT        25
#define MODEM_STATUS        34
#define LED_PIN             12
#define SerialMon Serial
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM7600  // Modem is SIM7600
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb

// Include after TinyGSM definitions
#include <TinyGsmClient.h>

const char apn[]      = "internet"; // Kullanilan servisin APN'i [Erişim Noktası Adı (Access Point Name)]
const char gprsUser[] = "vodafone"; // Kullanilan servis icin GPRS Kullanici adi
const char gprsPass[] = "vodafone"; // Kullanilan servis icin GPRS kullanici sifresi
const char simPIN[] = "";    // SIM card PIN code, if any

// MQTT Config
// EMQX Free Secure Broker for test https://www.emqx.io/mqtt/public-mqtt5-broker
// Check the website to find out more about the available options
// And obtain the current certificate
const char client_name[] = "MyEsp32";
const char mqtt_broker[] = "broker.emqx.io";
int secure_port = 8883; // TCP-TLS Port

// Layers stack
TinyGsm sim_modem(SerialAT);
TinyGsmClient gsm_transpor_layer(sim_modem);
SSLClient secure_presentation_layer(&gsm_transpor_layer);
PubSubClient client(secure_presentation_layer);

// For read the MQTT events
void callback(char *topic, uint8_t *payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++)
  {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

// To connect to the broker
[... 250 lines cut ...]
Modem Info: Manufacturer: SIMCOM INCORPORATED Model: SIMCOM_SIM7600G Revision: SIM7600M21-A_V2.0.1 SVN: 01 IMEI: 868822041837392 +GCAP: +CGSM
Waiting for network... OK
Connecting to network... OK
Connecting to APN: internet OK
CCID: **************
IMEI: **************
Operator: vodafone 
Local IP: **************
Signal quality: 26
Modem GPRS Check: TRUE
client.connected() FALSE
Attempting MQTT connection...failed, rc=-2...try again in 15 seconds
Attempting MQTT connection...[63361] ### Unhandled: +CIPCLOSE: 0,0
failed, rc=-2...try again in 15 seconds
Attempting MQTT connection...[79702] ### Unhandled: +CIPCLOSE: 0,0
failed, rc=-2...try again in 15 seconds
Attempting MQTT connection...[96847] ### Unhandled: +CIPCLOSE: 0,0
failed, rc=-2...try again in 15 seconds
```

**Environment, IDE:** Arduino IDE 2.3.2 with esp32 3.0.0 package (switched to 3.0.0 to solve a problem)  

Any help would be really appreciated, I'm stuck at connecting my SIM7600G to the Azure IoT for days. And if I manage to make this example work and then make Azure IoT work with this library a huge weight will be lifted on my back. 

Note: Been changing the code to try many things so if you see minor "strange" things in the code it's probably a leftover from these tries.

AltayAtaman, 2024-08-14:
I also tried to make it work with AT commands. Connecting to public brokers did work with following commands:

```
RDY

+CPIN: READY

SMS DONE

PB DONE
AT+CMQTTSTART
+CMQTTSTART: 0

AT+CCERTDOWN="ca_cert.pem",1358
>-----BEGIN CERTIFICATE-----
<certificate here, multiple lines>
-----END CERTIFICATE-----
OK

AT+CSSLCFG="sslversion",0,4
OK
AT+CSSLCFG="authmode",0,1
OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CMQTTSSLCFG=0,0
OK
AT+CMQTTREL=0
OK
AT+CMQTTACCQ=0,"Sim7600_fa7d2288",1
OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
OK

+CMQTTCONNECT: 0,0
```

But when I try to connecting it to Azure IoT Hub via AT commands, it doesn't connect instead gives +CMQTTCONNECT: 0,32 which is "handshake fail" according to the SIM7600 AT Command Manual. 

```

RDY

+CPIN: READY

SMS DONE

PB DONE
AT+CSSLCFG="sslversion",0,4
OK
AT+CSSLCFG="authmode",0,1
OK
AT+CSSLCFG="ignorelocaltime", 0, 1
OK
AT+CSSLCFG="ciphersuites", 0, 0xFFFF
OK
AT+CSSLCFG="cacert",0,"my_cert.pem"
OK
AT+CSSLCFG="enableSNI", 0, 1
OK
AT+CMQTTSTART
+CMQTTSTART: 0

OK
AT+CMQTTSSLCFG=0,0
OK
AT+CMQTTREL=0
OK
AT+CMQTTACCQ=0,"device_002",1
OK
AT+CMQTTCONNECT=0,"tcp://<my_broker>-devices.net:8883", 60, 1, "<my_broker>.azure-devices.net/device_002/?api-version=2021-04-12", "SharedAccessSignature sr=<my_broker>-devices.net%2Fdevices%2Fdevice_002&sig=********iS=********ZT=********Di********17246=********8"
OK

+CMQTTCONNECT: 0,32
```

Kostovite, 2025-05-23:
I having the similar problem with this model

RobertByrnes, 2026-11-02:
Please retest with GovoroxSSLClient/SSLClient 1.4.0: https://github.com/govorox/SSLClient/releases/tag/1.4.0

The current logs do not yet identify the failing layer:

- The printed rc=-2 comes from PubSubClient::state(), where it means MQTT_CONNECT_FAILED. It is not itself the SSLClient/mbedTLS error code.
- TinyGSM 0.11.7's SIM7600 client calls stop() before opening a connection, and stop() sends AT+CIPCLOSE. Therefore an Unhandled: +CIPCLOSE: 0,0 message can accompany cleanup; that line alone does not establish why TCP or TLS failed.
- The later AT+CMQTT / AT+CSSLCFG tests use the modem's built-in MQTT/TLS stack. The posted Arduino sketch instead uses PubSubClient -> SSLClient (TLS on the ESP32) -> TinyGsmClient (plain TCP). A result from one stack does not directly diagnose the other.

To isolate the failure, please test the public broker first, independently of Azure authentication:

1. Confirm registration and GPRS connectivity, then test a plain TinyGsmClient TCP connection to the same broker and port 8883. Stop that test connection before the next test. This checks DNS/TCP reachability only; it does not test TLS.
2. Test secure_presentation_layer.connect(mqtt_broker, secure_port) with the matching current CA and a correctly initialized ESP32 system clock. The posted sketch does not show any code synchronizing that clock. A modem clock update alone does not confirm that the ESP32 clock used for certificate validation is correct. Stop the standalone TLS test before testing PubSubClient.
3. If TCP/TLS succeed, test the MQTT connection and capture PubSubClient's result separately.

On failure, please capture the SSLClient error immediately, before another connection attempt:

```cpp
char errorText[128] = {};
int sslError = secure_presentation_layer.lastError(errorText, sizeof(errorText));
Serial.printf("SSLClient last error: %d (%s)\n", sslError, errorText);
```

Enable Arduino IDE Tools > Core Debug Level > Verbose and include logs from the start of the connection attempt through failure, along with the exact SSLClient, TinyGSM, PubSubClient, ESP32 core, and modem firmware versions. If using PlatformIO 1.4.0, use CORE_DEBUG_LEVEL=5 and SSL_CLIENT_LOG_LEVEL=5 as build flags. Please include relevant AT traffic if plain TCP fails, and redact tokens, credentials, and SIM/device identifiers.

For the separate Azure test, please verify that the broker hostname, MQTT username, and SAS token resource all refer to the same hub and device. The posted AT example uses <my_broker>-devices.net in the connection address/token resource but <my_broker>.azure-devices.net in the username. If that is not simply anonymization, correct the mismatch: Azure IoT Hub's hostname is <hub>.azure-devices.net. Its MQTT client ID must identify the registered device, with the appropriate username and SAS token (or supported certificate authentication). The anonymous public-broker client.connect(client_name) call is not the complete Azure authentication configuration. Do not post a live SAS token.

References: [PubSubClient error definitions](https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h), [TinyGSM 0.11.7 SIM7600 connect/stop handling](https://github.com/vshymanskyy/TinyGSM/blob/v0.11.7/src/TinyGsmClientSIM7600.h), and [Azure IoT Hub MQTT requirements](https://learn.microsoft.com/en-us/azure/iot-hub/iot-mqtt-connect-to-iot-hub).

Keeping this open with help wanted pending those results. We have not confirmed this SIM7600G setup is fixed by the newer release.