govorox/SSLClient #107: Reduce debug logging

Bascy, 2025-03-10:
When using this library with log level debug to allow our own debug messages to be logged, we get a lot of messages, and no way to disable those, except reduce the log level.

Can we move the log_d messages to log_v or use a #define to enable or disable debug logging all together?

RobertByrnes, 2026-11-01:
This is a valid enhancement request. On ESP32, the core already defines log_d/log_v, so the existing fallback LOG_LEVEL setting in SSLClient does not provide an independent cap for the library's messages.

I have opened PR #123 to add SSL_CLIENT_LOG_LEVEL, applied to SSLClient and certificate-bundle logs without redefining the application's logging macros: https://github.com/govorox/SSLClient/pull/123

For PlatformIO, the proposed configuration is:

```ini
build_flags =
  -D CORE_DEBUG_LEVEL=4
  -D SSL_CLIENT_LOG_LEVEL=2
```

This keeps the core/application debug level while limiting SSLClient to errors and warnings. Setting SSL_CLIENT_LOG_LEVEL=0 suppresses SSLClient's logs. The default is 5 to preserve existing behaviour; the core's own log level still limits output. Apply this as a build flag so it reaches the separately compiled library sources; a sketch-only define is insufficient.

The PR includes isolated logging checks and the existing native/example CI checks. It is not yet merged or released, so this option is not available in registry version 1.3.3. Keeping this issue open pending validation and release, and correcting the automated label to enhancement.

RobertByrnes, 2026-11-01:
Implemented in PR #123 and released in SSLClient 1.4.0: https://github.com/govorox/SSLClient/releases/tag/1.4.0

You can now keep application debug logging while limiting SSLClient to warnings and errors in PlatformIO:

```ini
build_flags =
  -D CORE_DEBUG_LEVEL=4
  -D SSL_CLIENT_LOG_LEVEL=2
```

Use SSL_CLIENT_LOG_LEVEL=0 to disable library logging. The setting covers SSLClient and certificate-bundle messages without changing application log macros. The default is 5 to preserve existing behaviour; the platform's own log level still applies. Apply the flag to library compilation, rather than defining it only in a sketch.

Version 1.4.0 has been published to PlatformIO; Arduino Library Manager will pick up the release through its indexing process. Logging checks, native tests, and Arduino/PlatformIO example builds passed before release.

Closing as completed. Thanks for requesting this improvement.

closed 2026-11-01 by RobertByrnes