govorox/SSLClient #113: LilyGO t-call Sim800L error -2

Bulbik1500, 2025-05-08:
Hello im using as in title lilygo t-call with build in sim800L(Firmware: Revision:1309B10SIM800L16 ) and i tried using:
https://github.com/govorox/SSLClient/blob/master/examples/Esp32-Arduino-IDE/https_gsm_SIM800/https_gsm_SIM800.ino
and im able to connet to network and get but when it comes to GET request i get Status code: -2

RobertByrnes, 2025-05-23:
Thanks for sharing the details of your setup and the issue. The status code -2 from SSLClient typically indicates a failure during the SSL/TLS connection or the HTTP request phase. Let's break down common causes and next steps:

### What Status Code -2 Means
- In SSLClient and similar Arduino libraries, status code -2 usually indicates a network or SSL handshake failure, not a standard HTTP error code.
- The error could come from:
  - Modem/network connectivity issues (but you said you can connect to the network, so less likely)
  - SIM800L not supporting certain SSL/TLS ciphers or being unable to establish a secure connection
  - Time not set correctly on the ESP32, which is required for certificate validation
  - Incorrect server address, port, or CA certificate
  - Power supply issues with the SIM800L during high current draw (e.g., during SSL handshake)

### Troubleshooting Steps

1. **Check Serial Output for Detailed Errors**
   - Look for more verbose logs in the Serial Monitor. Sometimes libraries print additional error info before returning -2.

2. **Check Time Synchronization**
   - SSL/TLS certificate validation requires the correct device time.
   - Ensure you synchronize the ESP32's time via NTP or GSM before making HTTPS requests.
   - Example for setting time (before HTTP request):
     ```cpp
     configTime(0, 0, "pool.ntp.org");
     while(time(nullptr) < SOME_REASONABLE_TIMESTAMP) { delay(100); }
     ```

3. **Check CA Certificate**
   - Make sure you are using the correct root CA certificate for the server you are connecting to.
   - If connecting to a public HTTPS site, extract and use the correct CA in PEM format.

4. **Server and URL**
   - Double-check the server URL (use `https`, correct port 443) and resource path.
   - Some sites (especially modern ones) require TLS 1.2 or higher, which SIM800L may not fully support.

5. **Power Supply**
   - SIM800L is very sensitive to voltage drops. Ensure you have a stable power supply capable of at least 2A during transmission peaks.

6. **Try HTTP Instead of HTTPS**
   - As a test, try switching to HTTP (if the server supports it) to see if the issue is specifically with SSL/TLS.

7. **SIM800L Firmware and Support**
   - SIM800L has limited and sometimes buggy SSL support. Consult the [SSLClient documentation](https://github.com/govorox/SSLClient) and SIM800L AT command manual for SSL limitations.

8. **Check Example Code**
   - Compare your code closely with the example. Sometimes a small difference (like missing CA cert, not waiting for time sync, etc.) can cause failures.

### Next Steps

- Can you share the full serial log output? This will help pinpoint where the -2 is coming from.
- Confirm if you set the system time before the HTTPS request.
- What server/URL are you trying to access?
- Have you tried with HTTP (port 80) for comparison?

If you provide the serial logs and clarify the above points might be able to help narrow it down

RobertByrnes, 2026-11-01:
A correction to my earlier reply: the reported `Status code: -2` comes from ArduinoHttpClient's `responseStatusCode()`, where it means `HTTP_ERROR_API` (the request is not in the expected state). It is not itself an SSLClient/mbedTLS error code. The linked SIM800 example currently ignores the return value from `get()`, so an earlier connection failure can be hidden by this subsequent -2.

Also, in this example the SIM800L supplies plain TCP through TinyGsmClient and SSLClient performs TLS on the ESP32. The modem's built-in TLS/cipher limitations are therefore not the explanation for this particular layering.

Please retest with GovoroxSSLClient/SSLClient 1.3.3 and replace the request/response section with the following so we can see the original failure:

```cpp
int requestResult = http_client.get("/a/check");
Serial.print("GET result: ");
Serial.println(requestResult);

if (requestResult != 0) {
  char errorText[128] = {};
  int sslError = secure_presentation_layer.lastError(errorText, sizeof(errorText));
  Serial.printf("SSLClient last error: %d (%s)\n", sslError, errorText);
} else {
  int statusCode = http_client.responseStatusCode();
  Serial.print("Status code: ");
  Serial.println(statusCode);
  if (statusCode > 0) {
    Serial.println(http_client.responseBody());
  }
}
http_client.stop();
```

Keep the existing GPRS cleanup after this section. Please share:

- The GET result and SSLClient last-error output, plus verbose connection logs (Arduino IDE: Tools > Core Debug Level > Verbose).
- ESP32 Arduino core, TinyGSM, ArduinoHttpClient, and SSLClient versions, and whether you changed the example's host or CA certificate.
- The ESP32's current UTC time immediately before connecting. The example does not initialize it. For cellular-only use, obtain valid time through a supported modem/network mechanism or another trusted clock and set the ESP32 clock with settimeofday(). The earlier Wi-Fi-style configTime() suggestion alone does not supply time synchronization through the TinyGSM modem transport, and setting the modem clock alone does not confirm the ESP32 clock is set.

Please redact credentials and SIM/device identifiers from logs. The example prints CCID and IMEI, which we do not need.

References: [ArduinoHttpClient error definitions](https://github.com/arduino-libraries/ArduinoHttpClient/blob/master/src/HttpClient.h), [request handling](https://github.com/arduino-libraries/ArduinoHttpClient/blob/master/src/HttpClient.cpp), and [ESP32 system time](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/system_time.html).

Keeping this open pending the diagnostic results and replacing the automated missing-template label with help wanted. We do not yet have enough evidence to identify a specific library defect.