govorox/SSLClient #114: How can the current time be set for certificate validation

moritz89, 2025-05-13:
- How does mbedtls get the current time to verify certificates?
  - I remember reading that the compile time is used?
- Does setting the time with `settimeofday` affect the time used by mbedtls to verify certificates validity?

RobertByrnes, 2025-05-23:
Hi, mbed-tls uses system time (time()) in embedded systems with linux or similar. However as ESP / Arduino etc is bare metal and doesn't have a real time clock we have to get the time from NTP or from GSM.

RobertByrnes, 2025-05-23:
Does setting the time with settimeofday affect the time used by mbedtls to verify certificates validity?

moritz89, 2025-05-23:
Great. So then the process of

1. connect to internet (LTE/WiFi)
2. sync time
3. connect to server

is the sensible steps to take. Would that mean setting time with settimeofday() far in the future past will cause the SSL/TLS handshake to fail?

RobertByrnes, 2026-11-01:
Yes, that is the right sequence: establish network connectivity, synchronize the ESP32 system clock, then open the TLS connection. If you already have a valid time from an RTC or another source, network access is not itself required to set the clock.

In normal ESP32 Arduino/ESP-IDF builds with mbedTLS certificate-date checking enabled, certificate validation uses the current system time, not the sketch compilation time. settimeofday() updates that system clock; use a correct UTC Unix timestamp. Updating a cellular modem clock alone does not automatically set the ESP32 clock.

With certificate verification enabled, setting the clock before a certificate's notBefore date or after its notAfter date should cause a new TLS handshake to fail certificate validation. The same applies to relevant certificates in the validated chain. This is not an immediate revalidation of an already-established connection. For a test, restore the correct clock afterwards and reconnect. Date checks do not apply when certificate verification is disabled with setInsecure(), or when the underlying mbedTLS build omits date checking.

A correction to my earlier reply: ESP32 does have an RTC timer and runs FreeRTOS in the normal Arduino environment. The important point is that its wall-clock time must be initialized correctly; it is not necessarily a valid current date after power-on.

References: [ESP-IDF system time](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/system_time.html) and [mbedTLS certificate-time checks](https://github.com/Mbed-TLS/mbedtls/blob/mbedtls-2.28.10/library/x509.c#L892).

Closing as answered and labelling this as a question. Please follow up if you observe different behaviour, with the ESP32 core version and certificate-verification configuration.

closed 2026-11-01 by RobertByrnes