govorox/SSLClient #115: W5500 permanent fix

MeY0u, 2025-05-19:
Hello,
Are there any plans to have a permanent fix for w5500 not working properly? Even the workaround doesn't help much.
Comparing performance of govorox between enc28j60 and w5500 is day and night.
Help please,

MeY0u, 2025-05-23:
Just run the library connecting to https server and enjoy the failing ride

RobertByrnes, 2026-11-01:
Version 1.3.3 now includes a receive-callback fix for the Ethernet idle-read behaviour behind the old W5500 workaround: a negative read while the underlying client is still connected returns MBEDTLS_ERR_SSL_WANT_READ instead of being treated as a fatal TLS error. This handling is active without the W5500_WORKAROUND define. See #120 and PR #121: https://github.com/govorox/SSLClient/pull/121

Please update to GovoroxSSLClient/SSLClient 1.3.3 and repeat your side-by-side HTTPS connection test. We have regression coverage for the receive behaviour and passing Arduino/PlatformIO example builds, but have not verified your W5500 hardware setup, so keeping this issue open pending your results.

If failures persist, please share the exact Ethernet library/version, a minimal sketch including the timeout calls, and verbose SSLClient connection logs with the last SSL error. setTimeout() takes milliseconds; setHandshakeTimeout() takes seconds. In 1.3.3, a positive setTimeout() value also overrides the handshake timeout at connect time, so please include both calls and their values if used. Please redact any credentials or private endpoint details.

Thanks for reporting the comparison with ENC28J60; an updated success-rate comparison on 1.3.3 would be useful.